libusb/libusb #412: threads_windows.c: Memory access violation

jite, 2018-03-27:
Commit 4be320236ec485dd32ada1b85d4c0551274d1f8a refactored the Windows locking handling. However when running pyusb (using libusb1 backend) on win10 (64bit/64bit Python) we ran into a problem while trying to find USB devices: it always gave an OSError exception: "access violation writing 0x0000000000000024". Worth noting is that it works fine to detect devices, however the object finalize/del stage fails.

Unsure if it's a problem in libusb or pyusb, but did some digging and found out that prior to the commit mentioned, it works fine. Anyway, what I suspect is causing the issue is that something references ```mutex``` after ```usbi_mutex_destroy()``` has been called (which runs ```DeleteCriticalSection(mutex)```). Quick test of adding a ```InitializeCriticalSection(mutex)```after the deletion eliminates the problem (while of course not a proper solution).

dickens, 2018-03-27:
Please provide a debug log of libusb.

JoshMayberry, 2018-06-13:
I also get this error when running the following code:

```
import wx
import usb

class MyForm(wx.Frame):
	def __init__(self):
		wx.Frame.__init__(self, None, wx.ID_ANY, "Example")
		panel = wx.Panel(self, wx.ID_ANY)
		sizer = wx.BoxSizer(wx.VERTICAL)

		button = wx.Button(panel, label="Run test()")
		button.Bind(wx.EVT_BUTTON, self.test)
		
		sizer.Add(button, 0, wx.ALL|wx.CENTER, 5)
		panel.SetSizer(sizer)

	def test(self, event):
		valueList = []
		devices = usb.core.find(find_all=True)
		for config in devices:
			valueList.append((config.idVendor, config.idProduct))
		print(valueList)

		event.Skip()

if __name__ == "__main__":
	app = wx.App(False)
	frame = MyForm()
	frame.Show()
	app.MainLoop()
```

To produce the error, press the button "Run test()" in the frame that appears after running the code.

Here is the error that I get:

```
Exception ignored in: <bound method _AutoFinalizedObjectBase.__del__ of <usb.backend.libusb1._Device object at 0x04307210>>
Traceback (most recent call last):
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\guest\AppData\Local\Programs\Python\Python36-32\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x00000014
^C
```

mcuee, 2018-06-13:
Please help to provide the debug log, setPYUSB_ERROR=debugand
LIBUSB_DEBUG=4 environment
variables and post the logs.
-- 
Xiaofan

JoshMayberry, 2018-06-14:
How do I set those variables? Also, where will the logs be generated to?

mcuee, 2018-06-14:
Okay, I can reproduce this issue myself by running your program.

```
(myenv) C:\work\pyusb\test>python wxusb.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000007ac] libusb: debug [libusb_init] created default context
[ 0.000000] [000007ac] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000000] [000007ac] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000000] [000007ac] libusb: debug [get_windows_version] Windows 11 or later 64-bit
[ 0.000000] [000007ac] libusb: debug [windows_init_clock] hires timer available (Frequency: 2734376 Hz)
[ 0.000000] [000007ac] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.000000] [000007ac] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [000007ac] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.000000] [000007ac] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.000000] [000007ac] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [000007ac] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [000007ac] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [000007ac] libusb: debug [windows_init] UsbDk backend is available
[ 0.000000] [000007ac] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [000007ac] libusb: debug [libusb_get_device_list]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [56]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5D]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5E]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [5F]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [60]
[ 0.015625] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [61]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.031249] [000007ac] libusb: debug [get_api_type] lower filter driver(s): bcbtums;btwampfl
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [1]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): libusb0
[ 0.031249] [000007ac] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [B]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): RtlWlanu
[ 0.031249] [000007ac] libusb: debug [get_api_type] upper filter driver(s): vwifibus
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [1B]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): CH341SER_A64
[ 0.031249] [000007ac] libusb: debug [get_api_type] upper filter driver(s): serenum
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [26]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.031249] [000007ac] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [3F]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.031249] [000007ac] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [43]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [44]
[ 0.031249] [000007ac] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.031249] [000007ac] libusb: debug [get_api_type] matched driver name against HID API
[ 0.031249] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.046875] [000007ac] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.046875] [000007ac] libusb: debug [winusb_get_device_list] allocating new device for session [46]
[ 0.046875] [000007ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_27CB&SUBSYS_2A8C103C&REV_01\3&11583659&0&EB' bus number 1
[ 0.046875] [000007ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_27C8&SUBSYS_2A8C103C&REV_01\3&11583659&0&E8' bus number 2
[ 0.046875] [000007ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_27C9&SUBSYS_2A8C103C&REV_01\3&11583659&0&E9' bus number 3
[ 0.046875] [000007ac] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1033&DEV_0194&SUBSYS_00000000&REV_03\FFFFFFFFFFFFFFFF00' bus number 4
[... cut ...]
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\work\pyenv64\myenv\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\work\pyenv64\myenv\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\work\pyenv64\myenv\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
```

JoshMayberry, 2018-06-14:
Awesome. I'm glad you were able to reproduce it.

jite, 2018-06-14:
It would be interesting to know later on if the debug logs actually helped you solve this particular issue or not (since you’ve been inclined not to touch this issue without one). I’m a developer myself and I know how frustrating it can be with bad bug reports so don’t get me wrong.

I did provide you with the commit that introduced the issue and a likely cause for it. It looked like it wouldn’t take the author more than a couple of hours tops to figure out what went wrong?

mcuee, 2018-06-15:
@jite  I can see what you come from. And thanks for reporting the issue and provide a possible cause which is great.

On the other hand, I have some issues with your comment. Firstly to ask for a debug log can usually help. Secondly you can see that "Priorities shifted at work right after this happened so couldn’t provide a log" and it only takes a few minutes to generate the debug log, not "a couple of hours". It can be difficult to find a couple of hours at your spare time working on open source projects like libusb. BTW, I do not know if it is really a couple of hours myself since I am not a programmer and I do not know much about the internal working mechanisms of libusb (mainly on technical support and testing side of the projects).

welitonfreitas, 2018-06-20:
same issue, any solutions or workaround?

JoshMayberry, 2018-07-03:
@mcuee @jite Have you found a way to patch this behavior?

jite, 2018-07-03:
@JoshMayberry I did an initial investigation of what caused it and found that the mutex variable is most likely referenced after it’s been destroyed. I don’t know the internals of libusb and I’m not that familiar with Windows threading/locks to find the exact culprit. I did a quick workaround by initializing the mutex right after it’s destroyed so that any action towards ’mutex’ would be valid (from memory perspective). It’s not a proper fix but it worked as a temporary workaround.

We’ve later seen other weird issues using libusb1 as backend (for a winusb device) to pyusb forcing us to move to libusb0-win32 backend instead. The big drawback doing so is that we need to install a device filter for our device pointing it to libusb0.

What kind of usb device do you have? Winusb?

JoshMayberry, 2018-07-05:
@jite The USB device in particular that I am using is a USB/COM barcode scanner. But, I get this issue regardless of it being plugged in or not. I hope that answered your question.
For now, I am using 'serial' to communicate with it.

mcuee, 2018-07-10:
@jite I see. Thanks for the clarifications. Hopefully this issue can be fixed (if it is confirmed to be an issue with libusb and not pyusb) once Chris or others have time to look into this.

mcuee, 2018-08-01:
Not so sure if this is related.
https://github.com/pyusb/pyusb/issues/186

mcuee, 2018-08-01:
Same as issue here in pyusb.
https://github.com/pyusb/pyusb/issues/203

mcuee, 2018-08-01:
Temporary workaround is to go back to libusb-1.0.21.

dickens, 2018-08-03:
```
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.673023] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C759548670 [2] (3.1)
[ 0.675556] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.682100] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7594D0830 [4] (3.0)
[ 0.684642] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.689348] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C759548670 [1] (3.1)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CC50>)
[ 0.692994] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7594709A0 [2] (3.3)
[ 0.695454] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.700620] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7594D0830 [3] (3.0)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CDD8>)
[ 0.706609] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7595492B0 [2] (2.2)
[ 0.708977] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CC50>)
[ 0.715358] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7594709A0 [1] (3.3)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.720169] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7594D0E50 [3] (2.0)
[ 0.722662] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CDD8>)
[ 0.728500] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7595492B0 [1] (2.2)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.733579] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C75946C990 [3] (2.1)
[ 0.736316] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.742019] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7594D0E50 [2] (2.0)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CC50>)
[ 0.746880] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C759548050 [3] (1.0)
[ 0.749325] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.755046] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C75946C990 [2] (2.1)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CDD8>)
[ 0.759850] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7594D0210 [4] (3.2)
[ 0.762281] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CC50>)
[ 0.768653] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C759548050 [2] (1.0)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.773529] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C7595498D0 [2] (3.4)
[ 0.776019] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CDD8>)
[ 0.782545] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7594D0210 [3] (3.2)
_Device initializer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.787474] [00004e20] libusb: debug [libusb_ref_device] ref device 000001C759548C90 [2] (1.1)
[ 0.789987] [00004e20] libusb: debug [libusb_get_device_descriptor]
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CBE0>)
[ 0.795905] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7595498D0 [1] (3.4)
_LibUSB finalizer (<usb.backend.libusb1._LibUSB object at 0x000001C75970CB70>)
[ 0.801349] [00004e20] libusb: debug [libusb_exit]
[ 0.803800] [00004e20] libusb: debug [libusb_exit] destroying default context
[ 0.806145] [00004e20] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 0.809560] [00004e20] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.813402] [000046c8] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
_Device finalizer (<usb.backend.libusb1._Device object at 0x000001C75970CCC0>)
[ 0.817313] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C759548C90 [1] (1.1)
[ 0.819770] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C759548670 [0] (3.1)
[ 0.823254] [00004e20] libusb: debug [libusb_unref_device] destroy device 3.1 (parent 000001C7594D0830)
[ 0.825610] [00004e20] libusb: debug [libusb_unref_device] unref device 000001C7594D0830 [2] (3.0)
Exception ignored in: <bound method _AutoFinalizedObjectBase.__del__ of <usb.backend.libusb1._DevIterator object at 0x000001C75970CBA8>>
Traceback (most recent call last):
  File "C:\Users\dickensc\AppData\Local\Programs\Python\Python36\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\dickensc\AppData\Local\Programs\Python\Python36\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
[... cut ...]
The refcounting mechanism in pyusb is broken.  The above log shows debug messages added within the ref/unref functions of libusb and the __init__/_finalize_object methods of pyusb.  You can see that the _LibUSB backend object was finalized before all of the _Device objects were finalized.  The warning from libusb during libusb_exit() even says _"some libusb_devices were leaked"_.  Applications are not allowed to reference a device after its context has been destroyed.

Closing this since this is not a libusb issue.

closed 2018-08-03 by dickens