libusb/libusb #415: Win32/minGW libusb_exit wait infinite never end

diabolo38, 2018-04-06:
under certain circumstance (gui app or not etc ..)  libusb_exit never end 
I found the  issue on win7-64 using MinGW32 static lib  

app get stuck on windows_destroy_clock on  WaitForSingleObject(timer_thread, INFINITE)
likely their is some racy or strange condition here for this logic to fail  ? 

libusb is linked as static lib inside my dll  used by a gui app 
the gui/app  does a first dll  load ask for device list (1 found) and then unload dll (no device open) it get stuck on the 1st libusb_exit.
this libusb_exit() call is done in context of "DLL main PROCESS DETACH" cause by gui app explicit dll unload (no end of process) 

debug output 
...
[ 0.133000] [00001a94] libusb: debug [libusb_exit]
[ 0.133000] [00001a94] libusb: debug [libusb_exit] destroying default context
[ 0.133000] [00001a94] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.133000] [00001270] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
=> inf wait ?! 

I instrumented and patched os  nt code as below to investigate and alo prevent unsafe infinite 

```
static void windows_destroy_clock(void)
{
    if (timer_thread) {
        HANDLE  th;
        DWORD tid;
        DWORD dwWait=-1;
        DWORD dwPost=-1;
        th = timer_thread;
        tid = timer_thread_id ;
        timer_thread_id =0;
        timer_thread = NULL; // try to prevent any racy re-use will killing it
              // actually the signal to quit the thread.
        // actually the signal to quit the thread.
        if (!(dwPost=pPostThreadMessageA(tid, WM_TIMER_EXIT, 0, 0))
                || ((dwWait=WaitForSingleObject(th, 500)) != WAIT_OBJECT_0)) {
            usbi_dbg("could not wait for timer thread to quit post %d wait=%d", (int)dwPost, (int)dwWait);
            TerminateThread(timer_thread, 1);
            // shouldn't happen, but we're destroying
            // all objects it might have held anyway.
        }
        CloseHandle(timer_thread);
    }

}
```
on 1st or later wanted exit (explciit dll unload) we get 
...
[ 0.142000] [00001644] libusb: debug [libusb_exit]
[ 0.142000] [00001644] libusb: debug [libusb_exit] destroying default context
[ 0.142000] [00001644] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.142000] [00001840] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
[ 0.642000] [00001644] libusb: debug [windows_destroy_clock] could not wait for timer thread to quit post 1 wait=258
[I] _usb_wrapper_over all done

=> wait exit after 500ms time out with code 258 = 0x102 ???  (ms say WAIT_TIMEOUT 0x00000102L)
 still thread exited 0.5 sec in past ??? how can  

on "GUI close"   or with console app close ouptut  is
DLL_PROCESS_DETACH
[ 3.246324] [00001c54] libusb: debug [libusb_exit]
[ 3.246324] [00001c54] libusb: debug [libusb_exit] destroying default context
[ 3.246324] [00001c54] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 3.246324] [00001c54] libusb: debug [windows_destroy_clock] could not wait for timer thread to quit post 0 wait=-1
[I] _usb_wrapper_over all done
=> different cause ?but we don't quite care no ?

I feel windows_destroy_clock  function shall be fixed in some way 
the simpler is to se finite time out (likley 100ms or ven less) to avoid app "freeze"  not nicest  fix :( 
Understanding and fixing  the real cause of freeze would be ideal  maybe theirs some racy condition over inf wait elsewhere?  I feel like debug print somehow race here as it test  first thread and then use thread_id  and if thread exit in meanwhile  will it ever get the event back ?

Note this issue is present on older version (1.0.21rc2 at least), i  only come to find it after adding missing "lib_usb exit" in my dll detach , that was causing libusb_init to fail after 47 dll unlaod/reload (timer thread create fail )

With  a "console" test application  this do not show it  seam to always show "could not wait for timer thread to quit"  wihtout any time out.

sorry I had not time to test on 64 bit nor win8/win10 version I may do next week

mcuee, 2018-06-28:
Do you still have the issue with libusb-1.0.22 release?

xm-repo, 2018-07-16:
hello, seems like for me its still there

diabolo38, 2018-08-03:
it was 1.0.22 (nano 11312)

dickens, 2020-01-14:
See [this](https://docs.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-best-practices) Microsoft document that describes some limitations of DllMain().  Because `libusb_init()` and `libusb_exit()` need to synchronize threads, these are unsafe to call from this context.

closed 2020-01-14 by dickens