libusb/libusb #419: Can't get libusb 1.0.22 working under Visual Studio 2017 (W10 1709)

C0rn3j, 2018-04-12:
This is probably user error.

Build log:
https://haste.c0rn3j.com/zetefudice.shell

Am trying to build for x64. I've downloaded the .7z file from releases, and am trying to follow the README.txt for 64 bit VS, static. 

I've unpacked the archive into C:\Users\0\source\repos\.

Am trying to compile this(mine) project (with two hacked modifications to even get it to attempt to compile under Windows) https://gitlab.com/C0rn3j/a4tech_bloody_p85_driver/

```
  - Copy libusb.h, from the include\libusb-1.0\ directory, into your project and
    make sure that the location where the file reside appears in the 
    'Additional Include Directories' section (Configuration Properties -> C/C++ -> General).
```
[image]

```
  - Copy the relevant .lib file from MS32\ or MS64\ and add 'libusb-1.0.lib' to
    your 'Additional Dependencies' (Configuration Properties -> Linker -> Input)
    Also make sure that the directory where libusb-1.0.lib resides is added to
    'Additional Library Directories' (Configuration Properties -> Linker
    -> General)
```
[image]
[image]

```
  - If you use the static version of the libusb library, make sure that
    'Runtime Library' is set to 'Multi-threaded DLL (/MD)' (Configuration
    Properties -> C/C++ -> Code Generation).
    NB: If your application requires /MT (Multi-threaded/libCMT), you need to
    recompile a static libusb 1.0 library from source.
```
[image]

```
  - Compile and run your application. If you use the DLL version of libusb-1.0,
    remember that you need to have a copy of the DLL either in the runtime
    directory or in system32
```
[image]

C0rn3j, 2018-04-12:
I've noticed in one of the issues that the mailing list is more appropriate, so I'll just close this here and link to it in my mail.

closed 2018-04-12 by C0rn3j

mcuee, 2018-04-21:
Thanks. Please refer to the following mailing list reply.
https://marc.info/?l=libusb-devel&m=152364219719888&w=2