libusb/libusb #420: LibUSB archive can't unpack in Windows

sambul13, 2018-04-15:
Several latest LibUSB.7z archives can't unpack in Windows 10 by latest 7z release. Error occurs when unpacking 32-bit fxload.exe .

Also, the documentation says nothing about installing LibUSB in Windows. I tried to install NetworkUPSTool, which is shipped with its own drivers, and it aborts install with error  "LibUSB missing. Install manually." But how exactly I can do it - not a word on Google or in this project. Why, if you provide Windows port?

diabolo38, 2018-04-16:
just copy libusb-1.0.dll (32 or 64 depending on app)  where is the application, you can aternatively create a folder somewhere copy the dll their and add this folder to the system path if you mix 32/64 bit version this wan't do for both :(

mcuee, 2018-04-16:
Which version of 7zip are you using? I have no such issues with 7zip at Windows 10. 

As for your issues with NUT, you can contact them to see they can help you. 

For libusb WIndows, please check the wiki.

sambul13, 2018-04-16:
@diabolo38 
Did you try that? When it says "LibUSB missing", it apparently means "no 3rd party USB Windows driver is installed for target device". How would I generate an INF from LibUSB folder with [INF-Wizard.exe](https://medium.com/@kirkbackus/creating-and-installing-an-inf-file-with-the-inf-wizard-caec58c3f7d5) shipped with libusbK-3.0.7.0-bin and libusb-win32-bin ?

@mcuee 
It seems irrelevant of 7z version, but somehow connected to Win Defender. It appears to check with a Cloud server if the file  fxload.exe is not a virus. I was able to unpack it only after rebooting Windows. It might also be related to conflict btw antimalware packages installed.

diabolo38, 2018-04-18:
If you never setup any back end driver then you can install one with libusk wizard or zadig

diabolo38, 2018-04-18:
Although you may face false alarm  virus detection and install problem with both of this that is what i fall into recently

mcuee, 2018-04-20:
Does NUT using libusb-1.0 API or libusb-0.1 API (libusb-win32)? If you refer to Inf-wizard, that is libusb-win32 which is a separate project from libusb.

mcuee, 2018-04-20:
https://github.com/networkupstools/nut/issues/300
So it seems NUT's main branch is still using libusb-0.1 API. I will close this. 

For NUT, please check with NUT people.
For libusb-win32, please ask in libusb-win32 mailing list (I am the admin there as well).
For 7zip apparently it is not a libusb problem.

closed 2018-04-20 by mcuee