libusb/libusb #421: libusb: error [udev_hotplug_event] ignoring udev action bind with BeagleBone Black

KarmaMan1982, 2018-04-20:
I've got a problem with this repository: https://github.com/ungureanuvladvictor/BBBlfs
It is useful to flash the BeagleBone Black - Onboard-Memory over USB!
Compiling works, but it isn't running since a few updates, the output:
root@TE-ENT21:/home/markus/BBBlfs/bin# ./usb_flasher 
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action unbind
SPL has started!

libusb: error [udev_hotplug_event] ignoring udev action bind
Cannot release interface!
----------------
uname -r: 4.15.0-15-generic
OS: Ubuntu 18.04 64 bit

If I do the same with pressing the S2-Button and then connecting to the USB-Port under Windows, a message shows that the device needs more power than the pc can support!

Please could you help me?
The original developer "ungureanuvladvictor" hasn't worked on his scripts since two years and don't reply to my messages!

Best regards Markus

mcuee, 2018-04-20:
Please use libusb mailing list for support.

diabolo38, 2018-04-20:
The solution is on the device side the config descriptor must be fixed  to reduce power  or even report self powered. I don t know how the bb gadget is setup  if with config fs the it shall be simple find the script and modify add it. If you search a bit theirs lot of help and sample how config fs work

mcuee, 2018-04-21:
To OP: if you use a power HUB, does the device report the same thing under Windows? 
I think this is not a libusb issue, but rather with the device or something else.

mcuee, 2018-06-09:
Close for now.

closed 2018-06-09 by mcuee

kczeladko, 2021-05-14:
device was connected via usb hub/switch , connecting directly to pc/notebook solve problem, thanks for help @mcuee