libusb/libusb #422: Cannot claim interface 1 on Windows, "Unsupported API"

kuro68k, 2018-04-26:
I'm trying to get dfu-util to work with a runtime DFU interface, which is interface 1 of a composite HID device. WCID has been used to load the WinUSB driver for this interface successfully.

You can see that only interface 0 (HID) seems to be supported. If I dig into the apib of interface 1 (DFU runtime) the designation is "Unsupported API".

```
priv->usb_interface
0x0085e008 {{path=0x0085f568 "\\\\.\\HID#VID_9999&PID_0008&MI_00#8&1054C979&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}" ...}, ...}
[0] {path=0x0085f568 "\\\\.\\HID#VID_9999&PID_0008&MI_00#8&1054C979&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}" ...}
[1] {path=0x00000000 <NULL> apib=0x00bb73e0 {dfu-util.exe!const windows_usb_api_backend usb_api_backend[5]} {...} ...}
[2] {path=0x00000000 <NULL> apib=0x00bb73e0 {dfu-util.exe!const windows_usb_api_backend usb_api_backend[5]} {...} ...}
[3] {path=0x00000000 <NULL> apib=0x00bb73e0 {dfu-util.exe!const windows_usb_api_backend usb_api_backend[5]} {...} ...}
[4] {path=0x00000000 <NULL> apib=0x00bb73e0 {dfu-util.exe!const windows_usb_api_backend usb_api_backend[5]} {...} ...}
```

I want to fix this myself but could do with some guidance. It looks like it is an issue with interface 1 not being supported by libusb when it should be. I'm using the WinUSB backend and having trouble finding where it decides if an interface is supported or not.

For reference here is the USB device I'm talking to:

https://github.com/kuro68k/xmega_usb

```

    =========================== USB Port2 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-1-2

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\.\usb#vid_9999&pid_0008#3555303335351909000b0#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Device ID                : USB\VID_9999&PID_0008\3555303335351909000B0
Hardware IDs             : USB\VID_9999&PID_0008&REV_0100 USB\VID_9999&PID_0008
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0057 (GUID_DEVCLASS_USB)
Driver Inf               : C:\Windows\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Interface GUID           : {a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0002.Hub_#0004
Location IDs             : PCIROOT(0)#PCI(1D00)#USBROOT(0)#USB(1)#USB(2)
Container ID             : {d63a0a4a-f231-54a6-8b00-6f67f7a8cb39}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 2
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : Runtime
  Device ID              : USB\VID_9999&PID_0008&MI_01\7&29E5561E&0&0001
  Problem                : 1 (CM_PROB_NOT_CONFIGURED)

        ---------------- Connection Information ---------------
Connection Index         : 0x02 (2)
Connection Status        : 0x01 (DeviceConnected)
Current Config Value     : 0x01
Device Address           : 0x02 (2)
Is Hub                   : 0x00 (no)
Number Of Open Pipes     : 0x01 (1)
Device Bus Speed         : 0x01 (Full-Speed)
Pipe0ScheduleOffset      : 0x00 (0)

        ------------------ Device Descriptor ------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
[... 106 lines cut ...]
        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01
bAlternateSetting        : 0x00
bNumEndpoints            : 0x00 (Default Control Pipe only)
bInterfaceClass          : 0xFE (Application Specific)
bInterfaceSubClass       : 0x01 (Device Firmware Upgrade)
bInterfaceProtocol       : 0x01 (Runtime)
iInterface               : 0x10 (String Descriptor 16)
 Language 0x0409         : "Runtime"

        ----- Device Firmware Update Functional Descriptor ----
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x21 (DFU Functional Descriptor)
bmAttributes             : 0x09
 Bit 7..4: reserved      : 0x00
 Bit 3: WillDetach       : 0x01
 Bit 2: ManifestTolerant : 0x00
 Bit 1: CanUpload        : 0x00
 Bit 0: CanDownload      : 0x01
wDetachTimeOut           : 0x0000 (0 ms)
wTransferSize            : 0x0200 (max 512 bytes)
bcdDFUVersion            : 0x0101 (Version 1.1)
```

kuro68k, 2018-04-26:
Okay, so it looks like the issue is that the device gets claimed during the DEV pass of winusb_get_device_list() because it matches the driver name against the composite API. As such it is then ignored in the GEN pass.

kuro68k, 2018-04-26:
Seems to be related to this: https://github.com/libusb/libusb/issues/370
And this: https://sourceforge.net/p/libusb/mailman/message/33176212/

So the issue is that the parent is using the composite driver, rather than WinUSB. I have confirmed this.

I am trying to get dfu-util to work on Windows with a runtime DFU interface. dfu-util looks for the DFU runtime descriptor in the device's configuration descriptor, and finds it in the parent composite device. I can't see how to make it find the child device, if that is at all possible.

mcuee, 2018-04-28:
Ref: https://github.com/libusb/libusb/issues/394
Ref: https://marc.info/?t=152481723000002&r=1&w=2

mcuee, 2018-04-28:
And to make sure it is not because of the default WinUSB driver using WCID, you can use Zadig to install the WinUSB driver again to see if that makes any differences. Last time I had issues before with ST-Link when using ST's driver but Zadig worked fine.

kuro68k, 2018-05-01:
I tested with DFU on interface zero, and got the same error.

I tested with a Zadig installed WinUSB driver and got the same error.

This seems to be a limitation of the WinUSB backend due to it not supporting the composite driver. I don't think there is any easy fix for composite devices in general, except manually replacing the default driver that Windows installs for them.

diabolo38, 2018-05-01:
It seam you will have anyway to use a custom inf and  back end driver. Why not to use libsubk or dk? For libusbk their is no issue I do use it with various composite device mix of standard class driver such as rndis CDC mass storage and vendor specific.

kuro68k, 2018-05-01:
The goal is to avoid using a custom INF. Aside from the issues with signing them and the like, the customers get confused and just want something that plugs in and works. I mean, people don't expect to need a driver for their keyboard...

I'm writing some test code now to see what other options are available.

diabolo38, 2018-05-01:
Make sense. Why not to have dfu device alone (not same vid pid) entered by s/w of normal function or mechaninal swicth at power up? What the benefit of having composite normal. + dfu function for  just f/w upgrade  ?

kuro68k, 2018-05-01:
That's the other option. It would be nice to be able to enter DFU via software though. For example I have some automated test equipment, and I'd like the host software to be able to update its firmware when required without user intervention.

I think it will have to be a custom solution though.

unnamalai-kb, 2018-09-14:
In the tests with DFU and WCID, WCID is sufficient to auto-load winusb.sys. However, dfu-util.exe was not able to open the DFU interface (at interface 1). It requires the DeviceInterfaceGUID regkey and this key is not added by Windows. When an additional DeviceInterfaceGUID descriptor is provided along with the WCID, then the dfu-util.exe could successfully locate the DFU interface. 

Please refer to https://github.com/pbatard/libwdi/wiki/WCID-Devices#Defining_a_Device_Interface_GUID_or_other_device_specific_properties

mcuee, 2020-08-31:
@kuro68k Can you try the latest git to see if the issue is still there or not.

kuro68k, 2020-09-02:
@mcuee I will try to find some time, as you might imagine I moved on to other projects but I can dig the hardware out again.

mcuee, 2020-09-03:
@kuro68k Thanks a lot for the help in advance.

mcuee, 2020-12-23:
One reference from FTDI. It uses the WinUSB WCID method to get Windows automatically use WinUSB as the driver.
https://www.ftdichip.com/Support/Documents/AppNotes/AN_344_FT51A_DFU_Sample.pdf

kuro68k, 2020-12-23:
I need to dig some development hardware out, it's all in boxes. But this was fixed, I had it working, I just can't remember what was done. It might have been the DeviceInterfaceGUID fix DeviceInterfaceGUID mentioned, or one of the two PRs I submitted to fix selection of interfaces.

I'll try to look at it over new year.

mcuee, 2021-01-25:
@kuro68k Thanks for the updates. I assume this is no longer an issue in the latest git, right? If that is the case, I will close this issue.

kuro68k, 2021-01-25:
Sorry I've been really busy and not had time to check it.

mcuee, 2021-07-05:
@kuro68k Close this issue for now. Please reopen if you have new info.

closed 2021-07-05 by mcuee

sresam89, 2022-06-09:
Sorry to bump an old thread

> When an additional DeviceInterfaceGUID descriptor is provided along with the WCID, then the dfu-util.exe could successfully locate the DFU interface.

@unnamalai-kb could you explain in detail how you did this? I mean the how the additional parameters were given to dfu-util