libusb/libusb #424: Windows download is a 7zip file

cellularmitosis, 2018-04-29:
The "Latest Windows binaries" link is a 7-zip file.  Can you make this a plain zip file instead?

mcuee, 2018-05-13:
Why? I usually install Google Chrome 7-zip and Notepad++ and a few software after installation of Windows.

cellularmitosis, 2018-05-15:
That's great for you.  But that doesn't change the fact that you are forcing literally everyone who downloads this to also install 7-zip.  When you are hosting something for other people, usually you want to cater to their needs, not your own.

Grix, 2018-05-27:
>When you are hosting something for other people, usually you want to cater to their needs, not your own.

Hm, remember that libusb is provided free of charge and restrictions out of the goodwill of the authors. Its users should be grateful, not entitled. I mean it's obviously not a huge issue in this case, but it's the principle. 

Also the fact is the 7z offers a lot better compression which results in lower hosting costs for the hoster, and faster downloads for users. And I would find it very strange if any developers do not have proper decompression software installed already.

mcuee, 2018-06-14:
Close this for now.

closed 2018-06-14 by mcuee

wez, 2019-07-16:
I'd like to second the request to add a zip file download alternative.  In my situation I can easily programmatically extract a zip file from rust.  A 7z crate does not currently exist for rust and the effort to build that library is much higher than adding a zip file download to this project.

elfmimi, 2022-01-01:
One workaround for this problem is to grab it from MSYS2's package.
https://packages.msys2.org/package/mingw-w64-x86_64-libusb
- Unfortunately you'll need [zstd](https://github.com/facebook/zstd/releases/) and `tar` to extract.