libusb/libusb #436: Crash using Logitech Vantage Mic (Rock Band) with USB Passthrough Feature (Isochronous Transfer) of Dolphin Emulator

PipeRojas, 2018-05-28:
Hi everybody, I'm having an issue, I'm trying to play guitar hero and rock band on dolphin emulator with the USB Logitech vantage microphone with the USB passthrough feature and usbdk 1.00-19 but after registering the device in the white list of dolphin usb passthrough and starting the game, it crashes randomly, sometimes I can play a half of a song and dolphin crashes with Windows 10 Pro x64 up to date (fall creators update).
Anybody knows any solution?

And thanks a lot btw

PipeRojas, 2018-05-29:
https://1drv.ms/u/s!Amvh0udatWAwiFw54anpi7CadZ1J

Here´s a dump file of Dolphin just before it crashes.

Tnx for the help

mcuee, 2018-06-11:
Just wondering if you use libusbK backend, do you encounter this? Take note libusb now supports isoc transfer under Windows (1.0.22 onwards) with the libusbK backend. 

usbdk has the advantage though since you can keep the original driver. But for debugging purpose, you may try libusbk first to see if that helps.

sameehj, 2018-06-11:
@PipeRojas How did you generate this dump file? I need the one generated by Windows when the BSOD occurs. It usually resides in C:/windows/memory.dmp

I can't UsbDk in the stack inside the dump that you've sent

PipeRojas, 2018-06-12:
@sameehj I have not the bsod dump file cuz when the crash appears, it won't generate a bsod, instead, the emulator just crashes and it won't generate the "Dolphin stopped working", it just crashes and goes directly to desktop. Is there other way to capture a dump file when applications just get closed when crash appears??
Thanks btw

PipeRojas, 2018-06-13:
@mcuee I tried libusbk with the same results, the emulator just closes a few seconds after starting any RockBand or GuitarHero game, without bsod screen or Stopped Working dialog. Sorry about my ignorance but idk how to debug the driver.

Tnx for the help

mcuee, 2018-06-13:
If this is the case it has nothing to do with usbdk driver or libusbk
driver, either Dolphin is the culprit or isoc transfer support inside
libusb (either the usbdk backend or the libusbk backend) is not working for
the use of Dolphin emulator.

BTW, make sure you use the latest version of libusb — you need that to have
the isoc support for libusbk driver. You will need to enable usbdk support
by using libusb_set_option() in this case.
-- 
Xiaofan

mcuee, 2018-06-13:
You may also ask in the Dolphin support channel to see it is a general
problem or specific to you.
-- 
Xiaofan

PipeRojas, 2018-06-13:
Thanks for the answer,

I´ve been asking in the Dolphin Forums and I realized that the problem only occurs on Windows. Linux can handle this kind of transfer mode but its problematic on Windows, even more on Windows 10. So off course i tried using Ubuntu, but the OpenGL overall performance is low than Direct3D. Thats why im looking for a solution using Windows 10.

Besides, the USB Passthrough feature of Dolphin only works on windows with usbdk or libusbk installed, and i think that when a crash with dolphin is caused by itself, will generate an "Stopped Working" dialog, but when the emu crashes by this issue, simply it closes and the USB plugged device sound of windows occurs (Coming back to windows driver, i think)

Tnx btw

mcuee, 2018-06-14:
You may want to set LIBUSB_DEBUG=4 and then run post the debug log. The issue may have nothing to do with libusb -- it could be your device problem or the Dolphin problem.

When you say hear the Windows Device Sound, did you get it when using libusbK? You should not. It could be your device is faulty.

PipeRojas, 2018-06-14:
Hi,

No, ive been verifying that and windows sound occurs using usbdk, but both backends will crash. I will try to debug the driver behavior, tnx.

mcuee, 2018-06-14:
That is better. Sometimes you will hear the Windows sound with usbdk backend. You need to post the debug log by setting LIBUSB_DEBUG=4. I think this has nothing to do with the driver (libusbK / usbdk), rather it is either the problem of Dolphin or the problem with libusb library's isoc support. I suspect it is Dolphin.

PipeRojas, 2018-06-14:
I've found the corresponding issue in Dolphin forums: https://bugs.dolphin-emu.org/issues/10141 
I will try to debug and post the log anyways.

mcuee, 2018-06-15:
Thanks. I think your issues are related to #232.

mcuee, 2018-06-15:
#232

Hacsev, 2019-02-15:
Will this issue ever be addressed and fixed? All I have seen is a back and forth of "this is X repo's problem". If it is, can you point them in the right direction to have this issue finally fixed? I'm sure a lot of Rock Band players will be grateful to finally have mics working.

mcuee, 2020-08-31:
What if you use WinUSB driver and not usbdk driver? Does it work?

PipeRojas, 2020-08-31:
Hi everyone, I found out that the issue was solved in Dolphin Emulator and I´ve only needed to update my emulator version, as you can see here https://wiki.dolphin-emu.org/index.php?title=USB_Passthrough in the Windows section, where is specified the exact version to avoid the issue, im closing this issue and sorry about that.

closed 2020-08-31 by PipeRojas

mcuee, 2020-09-01:
Great. Thanks for the updates.

Based on the Dolphin Wiki, usbdk is not recommended to be used.