libusb/libusb #437: libusb_control_transfer error

cdedwards, 2018-06-01:
I'm attempting to port a device driver from linuxcnc to Mach4 cnc controller for the XHC04 wireless hand controller.  This device works unders Mach3 quite well and I have to assume it works on linuxcnc as it is actively being modified. data is a 48 byte char[]. it's divided into 6 packets of 8 bytes for the device. I get a error 1 every time using this method. This is a direct copy from the linux source with the added debugging for windows. the failure is on a DeviceIoControl(). Error code is Incorrect Function.

		int r = libusb_control_transfer(xh->dev_handle,
			LIBUSB_DT_HID, //bmRequestType 0x21
			LIBUSB_REQUEST_SET_CONFIGURATION, //bRequest 0x09
			0x0306,         //wValue
			0x00,           //wIndex
			data + 8 * packet,  //*data
			8,           //wLength
			0);             //timeout
		if (r < 0) {
			r = GetLastError();
			sprintf(buf, "XHC Error %d %s", r, hexStr(data, 8));
			XHC_DEBUG(buf);
			xh->do_reconnect = 1;
		}

mcuee, 2018-06-09:
You may need to provide the error log. And it seems that you are using an HID device. Are you using the Windows default HID driver or WinUSB?

mcuee, 2018-06-14:
You can set LIBUSB_DEBUG=4 and the post the debug log.

2ndTaleStudio, 2018-09-14:
Hi there,
may I just continue on this issue as my problem might be related. I try to use libusb on Win10 to access a PS3 controller over USB. I've been using the pre-compiled Win10 version and run the xusb test programm with option -p and -d.
While reading some master data seem to work fine and the controller is identified the reading of the BD master address or device address failes...
The debugging shows this:

`[ 0.326992] [000036c4] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.327992] [000036c4] libusb: debug [hid_submit_control_transfer] will use interface 0
[ 0.328991] [000036c4] libusb: debug [_hid_get_descriptor] LIBUSB_DT_STRING
[ 0.329991] [000036c4] libusb: debug [libusb_free_transfer] transfer 00000000012DC328
[ 0.331992] [000036c4] libusb: debug [libusb_alloc_transfer] transfer 00000000012DC328
[ 0.332992] [000036c4] libusb: debug [libusb_submit_transfer] transfer 00000000012DC328
[ 0.333991] [000036c4] libusb: debug [hid_submit_control_transfer] will use interface 0
[ 0.334994] [000036c4] libusb: debug [_hid_get_report] report ID: 0xF5
[ 0.335992] [000036c4] libusb: debug [_hid_get_report] Failed to Read HID Report: [87] The parameter is incorrect.
[ 0.336990] [000036c4] libusb: debug [libusb_free_transfer] transfer 00000000012DC328
   Input/Output Error`

The relevant coding in the source code is:
`// Get the controller's bluetooth address of its master device
	CALL_CHECK(libusb_control_transfer(handle, LIBUSB_ENDPOINT_IN|LIBUSB_REQUEST_TYPE_CLASS|LIBUSB_RECIPIENT_INTERFACE,
		HID_GET_REPORT, 0x03f5, 0, master_bt_address, sizeof(master_bt_address), 100));`

The call is passing **0x03F5** as wValue but the debug output indicates that the reportId **0xF5** is invalid. Could it be that the **0x300** is somehow stripped away from this value? I've tested on 2 machines with Win10 and the latest dll version of the lib...

Any hint's where welcome and much appreciated...

lixikun, 2019-06-26:
hi,how to deal with the error code : 87 ?

mcuee, 2021-01-23:
Please use libusb mailing list for support. Close this for now.

closed 2021-01-23 by mcuee