libusb/libusb #441: Windows 7, when USB3.0 root hub doesn't have child device, libusb_device_list will list nothing

changyp6, 2018-06-06:
I recently encountered a very odd error.
The windows 7 system has a driver issue, which USB3.0 driver cannot work properly. The USB hub in the windows device manager is always "Unknown Device".
When I call libusb_device_list(), nothing is reported, even if I connect my device on USB2.0 port.

After I enable the debug log, libusb_device_list() reports:
**program asserrtion failed - HCD "HCD\VEN_8086&DEV_xxxx" child not found**

However, the old libusb library, which is **libusb 1.0.21 can work well** on this Windows system.

I think one broken USB hub, should not block other working USB hub.

Please help take a look at this issue.

mcuee, 2018-06-11:
Please provide the debug message (eg: using xusb -d", or set LIBUSB_DEBUG=4 and then run listdev example). Thanks.

changyp6, 2018-07-05:
@mcuee Sorry for the delay. I just saw your reply.

**Following is the debug output from listdevs.exe**
` D:\libusb-1.0.22-win64-debug>listdevs.exe
 [timestamp] [threadID] facility level [function call] <message>
 --------------------------------------------------------------------------------
 [ 0.000000] [000015ec] libusb: debug [libusb_init] created default context
 [ 0.000000] [000015ec] libusb: debug [libusb_init] libusb v1.0.22.11312
 [ 0.010000] [000015ec] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
 [ 0.010000] [000015ec] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
 [ 0.010000] [000015ec] libusb: debug [windows_init_clock] hires timer available (Frequency: 3312871 Hz) 
 [ 0.010000] [000015ec] libusb: debug [windows_init_clock] timer thread will run on core #0
 [ 0.020000] [000015ec] libusb: debug [htab_create] using 1021 entries hash table
 [ 0.020000] [000015ec] libusb: debug [winusbx_init] using libusbK DLL for universal access
 [ 0.020000] [000015ec] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
 [ 0.030001] [000015ec] libusb: debug [winusbx_init] initalized sub API libusbK
 [ 0.030001] [000015ec] libusb: debug [winusbx_init] initalized sub API libusb0
 [ 0.030001] [000015ec] libusb: debug [winusbx_init] initalized sub API WinUSB
 [ 0.040001] [000015ec] libusb: info [windows_init] UsbDk backend is not available
 [ 0.040001] [000015ec] libusb: debug [usbi_add_pollfd] add fd 0 events 1
 [ 0.040001] [000015ec] libusb: debug [libusb_get_device_list]
 [ 0.410016] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [23C8]
 [ 0.410016] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [2404]
 [ 0.410016] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [2440]
 [ 0.420016] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [2490]
 [ 0.420016] [000015ec] libusb: debug [get_api_type] driver(s): libusb0
 [ 0.420016] [000015ec] libusb: debug [get_api_type] matched driver name against libusb0
 [ 0.430017] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [8A8]
 [ 0.430017] [000015ec] libusb: debug [get_api_type] driver(s): usbccgp
 [ 0.430017] [000015ec] libusb: debug [get_api_type] matched driver name against Composite API 
 [ 0.430017] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [1DA0]
 [ 0.440017] [000015ec] libusb: debug [get_api_type] driver(s): HidUsb
 [ 0.440017] [000015ec] libusb: debug [get_api_type] matched driver name against HID API 
 [ 0.440017] [000015ec] libusb: debug [winusb_get_device_list] allocating new device for session [1DF0]
 [ 0.450018] [000015ec] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1E26&SUBSYS_052C1028&REV_04\3&11583659&0&E8' bus number 1
 [ 0.450018] [000015ec] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1E2D&SUBSYS_052C1028&REV_04\3&11583659&0&D0' bus number 2
 [ 0.450018] [000015ec] libusb: error [enumerate_hcd_root_hub] program assertion failed - HCD 'PCI\VEN_8086&DEV_1E31&SUBSYS_052C1028&REV_04\3&11583659&0&A0' child not found
 [ 0.460018] [000015ec] libusb: debug [libusb_unref_device] destroy device 1.0 
 [ 0.460018] [000015ec] libusb: debug [libusb_unref_device] destroy device 2.0 
 [ 0.460018] [000015ec] libusb: debug [libusb_unref_device] destroy device 0.0 
 [ 0.460018] [000015ec] libusb: debug [libusb_unref_device] destroy device 0.0 
 [ 0.470018] [000015ec] libusb: debug [libusb_unref_device] destroy device 0.0 
 [ 0.470018] [000015ec] libusb: debug [libusb_unref_device] destroy device 0.0 
 [ 0.470018] [000015ec] libusb: debug [libusb_unref_device] destroy device 0.0`

**Following is the debug output from xusb.exe -d**
`D:\libusb-1.0.22-win64-debug>xusb.exe -d
Using libusb v1.0.22.11312

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00001994] libusb: debug [libusb_init] created default context
[ 0.010000] [00001994] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.010000] [00001994] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.020000] [00001994] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.020000] [00001994] libusb: debug [windows_init_clock] hires timer available (Frequency: 3312871 Hz)
[ 0.020000] [00001994] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.030001] [00001994] libusb: debug [htab_create] using 1021 entries hash table
[ 0.030001] [00001994] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.030001] [00001994] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.040001] [00001994] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.040001] [00001994] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.050002] [00001994] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.050002] [00001994] libusb: info [windows_init] UsbDk backend is not available
[ 0.050002] [00001994] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 0000:0000...
[ 0.060002] [00001994] libusb: debug [libusb_get_device_list]
[ 0.370014] [00001994] libusb: debug [winusb_get_device_list] allocating new device for session [23C8]
[... cut ...]
[ 0.430017] [00001994] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.430017] [00001994] libusb: debug [libusb_unref_device] destroy device 0.0
[ 0.440017] [00001994] libusb: debug [libusb_unref_device] destroy device 0.0
  Failed.
[ 0.440017] [00001994] libusb: debug [libusb_exit]
[ 0.440017] [00001994] libusb: debug [libusb_exit] destroying default context
[ 0.450018] [00001994] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.450018] [00001a1c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting`

changyp6, 2018-07-05:
@mcuee as you can see from the log, if `libusb: error [enumerate_hcd_root_hub] program assertion failed - HCD 'PCI\VEN_8086&DEV_1E31&SUBSYS_052C1028&REV_04\3&11583659&0&A0' child not found` is printed, libusb exited instantly, child devices under "bus 1 and bus 2" are ignored without listing.

jiez, 2018-09-24:
@changyp6 have you tried to use "git bisect" to narrow it down to a specific commit?

changyp6, 2018-09-25:
> @changyp6 have you tried to use "git bisect" to narrow it down to a specific commit?

The quick fix should be:
windows_winusb.c
insert the following code after line 1447
if (r == LIBUSB_ERROR_NO_DEVICE) {
    usb_warn(ctx, "Cannot find child device, driver bug?\n");
    r = LIBUSB_SUCCESS;
}

But I don't know if @mcuee has better solution

jiez, 2018-09-26:
@changyp6 Maybe you can take a look at enumerate_hcd_root_hub to find out why it returns LIBUSB_ERROR_NO_DEVICE on your machine. That function was part of the heavy rewrite 71a779d0 .

changyp6, 2018-09-26:
@jiez Like I said in the first message, it is caused by driver issue. Older libusb can handle such driver issue.

aksonlyaks, 2019-01-26:
Dear all,

how to resolve this issue? As I'm facing this issue in two of my laptops which has USB 3.0 port along with USB 2.0 port.

Thanks

jiez, 2019-01-26:
See the pull request above.

aksonlyaks, 2019-01-31:
Is there any plan to release these changes? Or release any pre-releases? Since Mar-18 there is no pre-release or release.

dickens, 2020-01-03:
I would like to understand this issue further.  I agree that failure to enumerate a host controller's root hub should not stop the entire enumeration process, so I will merge your PR #483.  I'm curious what the value of `child_devinst` is for the failing root hub.  Would you be able to add a debug message to print that out and provide the log?  Thanks!

commit a36862b1 (2020-01-03): windows: relax HCD root hub enumeration errors

Enumeration errors happen on one HCD root hub should not prevent
enumerating other hubs. This will fix an issue seen on some Windows 7
hosts with USB 3.0 host controller.

Closes #441, Closes #483
closed 2020-01-03 by dickens