libusb/libusb #442: Windows enumeration: configuration descriptor info

mcuee, 2018-06-11:
Not really a big issue, just want to point out that the following two devices emit some debug info during the enumeration.

1A40:0201 is a powered hub
04d8:0033 is Microchip PICkit 2 which has two configurations (HID and a custom configuration).

I think it is Windows limitation for the Microchip PICkit 2 and the 2nd configuration is not really needed. USBView also does not list down the configuration descriptor for the 2nd configuration.

For 1A40:0201, USBView can read the configuration descriptor

`
c:\work\libusb\bin-1.0.22\examples\bin64>xusb 0403:6001
Using libusb v1.0.22.11312

Opening device 0403:6001...
libusb: info [cache_config_descriptors] could not access configuration descriptor 0 (dummy) for 'USB\VID_1A40&PID_0201\6&142A28FA&0&5': [31] A device attached to the system is not functioning.
libusb: info [cache_config_descriptors] could not access configuration descriptor 1 (dummy) for 'USB\VID_04D8&PID_0033\PK2NEW': [31] A device attached to the system is not functioning.
  Failed.

c:\work\libusb\bin-1.0.22\examples\bin64>xusb -d 0403:6001
Using libusb v1.0.22.11312

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00001548] libusb: debug [libusb_init] created default context
[ 0.000000] [00001548] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000000] [00001548] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000000] [00001548] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000000] [00001548] libusb: debug [windows_init_clock] hires timer available (Frequency: 2734376 Hz)
[ 0.000000] [00001548] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.000000] [00001548] libusb: debug [htab_create] using 1021 entries hash table
[ 0.015625] [00001548] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.015625] [00001548] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.015625] [00001548] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.015625] [00001548] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.015625] [00001548] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.015625] [00001548] libusb: debug [windows_init] UsbDk backend is available
[ 0.015625] [00001548] libusb: debug [usbi_add_pollfd] add fd 0 events 1
Opening device 0403:6001...
[ 0.031249] [00001548] libusb: debug [libusb_get_device_list]
[ 0.031249] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [55]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [56]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [5D]
[ 0.046875] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [5E]
[ 0.046875] [00001548] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.046875] [00001548] libusb: debug [get_api_type] lower filter driver(s): bcbtums;btwampfl
[ 0.062499] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [1]
[ 0.062499] [00001548] libusb: debug [get_api_type] driver(s): libusb0
[ 0.062499] [00001548] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.062499] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [A]
[ 0.062499] [00001548] libusb: debug [get_api_type] driver(s): FTDIBUS
[ 0.062499] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [B]
[ 0.062499] [00001548] libusb: debug [get_api_type] driver(s): RtlWlanu
[ 0.062499] [00001548] libusb: debug [get_api_type] upper filter driver(s): vwifibus
[ 0.062499] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [19]
[ 0.062499] [00001548] libusb: debug [get_api_type] driver(s): CH341SER_A64
[ 0.062499] [00001548] libusb: debug [get_api_type] upper filter driver(s): serenum
[ 0.062499] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [25]
[ 0.062499] [00001548] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.062499] [00001548] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.078123] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [40]
[ 0.078123] [00001548] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.078123] [00001548] libusb: debug [winusb_get_device_list] allocating new device for session [41]
[... 124 lines cut ...]
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.3
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.6
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.7
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.5
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 5.0
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.4
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.5
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.6
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.2
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.1
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.1
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 3.0
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 4.0
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.8
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.2
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 6.0
[ 0.282187] [00001548] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.297812] [00001548] libusb: debug [libusb_exit]
[ 0.297812] [00001548] libusb: debug [libusb_exit] destroying default context
[ 0.297812] [00001548] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.297812] [000032c8] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
`

mcuee, 2018-06-11:
usbview output for the hub.

```
External Hub: USB#VID_0BDA&PID_5411#6&38fe5f06&0&3#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Hub Power:               Self Power
Number of Ports:         4
Power switching:         Individual
Compound device:         No
Over-current Protection: Individual

Device Descriptor:
bcdUSB:             0x0210
bDeviceClass:         0x09
bDeviceSubClass:      0x00
bDeviceProtocol:      0x02
bMaxPacketSize0:      0x40 (64)
idVendor:           0x0BDA (Realtek Semiconductor Corp.)
idProduct:          0x5411
bcdDevice:          0x0117
iManufacturer:        0x01
iProduct:             0x02
iSerialNumber:        0x00
bNumConfigurations:   0x01

ConnectionStatus: DeviceConnected
Current Config Value: 0x01
Device Bus Speed:     High
Device Address:       0x03
Open Pipes:              1

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0001 (1)
bInterval:            0x0C

Configuration Descriptor:
wTotalLength:       0x0029
bNumInterfaces:       0x01
bConfigurationValue:  0x01
iConfiguration:       0x00
bmAttributes:         0xE0 (Bus Powered Self Powered Remote Wakeup)
MaxPower:             0x00 (0 Ma)

Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x00
bNumEndpoints:        0x01
bInterfaceClass:      0x09 (Hub)
bInterfaceSubClass:   0x00
bInterfaceProtocol:   0x01
iInterface:           0x00

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0001 (1)
bInterval:            0x0C

Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x01
bNumEndpoints:        0x01
bInterfaceClass:      0x09 (Hub)
bInterfaceSubClass:   0x00
bInterfaceProtocol:   0x02
iInterface:           0x00

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0001 (1)
bInterval:            0x0C

```

dickens, 2020-08-17:
I think such errors result from the device not responding correctly to a descriptor request.  Since this the hub is using the Microsoft-provided driver, the issue is almost certainly going to be with the device.

As I don't think there's any recourse for libusb, I'm going to close this issue.  Feel free to re-open if new information comes up.

closed 2020-08-17 by dickens