libusb/libusb #444: libusb_init() return -99 in Android5.1.1, but return 0 in Android4.4.2

Shuyun6, 2018-06-12:
Code like : `int res = libusb_init(NULL);`
am I using a wrong method? or libusb cannot support Android 5.+?

mcuee, 2018-06-12:
Duplicate: will close this one.
https://github.com/libusb/libusb/issues/233
https://github.com/libusb/libusb/issues/188

You can try pull request.
https://github.com/libusb/libusb/pull/242

closed 2018-06-12 by mcuee

Shuyun6, 2018-06-12:
@mcuee it means I should invoke `libusb_wrap_fd()` before `libusb_init()`, right?

Shuyun6, 2018-06-13:
@mcuee [#242](https://github.com/libusb/libusb/pull/242) means that use `libusb_wrap_fd()` instead `libusb_open()`, but no anything about `libusb_init()`. I use [libuvc](https://github.com/ktossell/libuvc) that build atop libusb, it initilized through `libusb_init()`. So, how can I use `libusb_init()`?

mcuee, 2018-06-13:
It means currently unmodified libusb will not work for you — you need to
apply the patches provided by that pull request first.
-- 
Xiaofan