libusb/libusb #445: libusb_claim_interface on windows 10 LIBUSB_ERROR_NOT_SUPPORTED = -12,

guanguanboy, 2018-06-14:
does this issue have a solution which would worked out? I got this err /** Operation not supported or unimplemented on this platform */
LIBUSB_ERROR_NOT_SUPPORTED = -12,
when calling libusb_claim_interface on windows 10 with WinUsb installed using zadig

mcuee, 2018-06-14:
You need to tell us what you are doing. 
If you need technical support, please use libusb mailing list. Github issues are more meant for reporting bugs of libusb. Thanks.

mcuee, 2018-06-14:
BTW, I deleted your other entry since your question has nothing to do with that discussions.

guanguanboy, 2018-06-14:
I am testing  libuvc to capture video frames from a 3d device on windows, I simple run the example.c from the libuvc sourcecode on visual studio2013.
 when starting the stream, the program run into the below function, and reported -12 after calling  libusb_claim_interface(devh->usb_devh, idx);

here is the libuvc function:
uvc_error_t uvc_claim_if(uvc_device_handle_t *devh, int idx) {
  int ret;

  UVC_ENTER();

  /* Tell libusb to detach any active kernel drivers. libusb will keep track of whether
   * it found a kernel driver for this interface. */
  ret = libusb_detach_kernel_driver(devh->usb_devh, idx);

  if (ret == UVC_SUCCESS || ret == LIBUSB_ERROR_NOT_FOUND || ret == LIBUSB_ERROR_NOT_SUPPORTED) {
    UVC_DEBUG("claiming interface %d", idx);
    ret = libusb_claim_interface(devh->usb_devh, idx);
  } else {
    UVC_DEBUG("not claiming interface %d: unable to detach kernel driver (%s)",
              idx, uvc_strerror(ret));
  }

  UVC_EXIT(ret);
  return ret;
}

after this error occured, I just run the testlibusb.c(from the libusb sourcecode in github and i added this line of code: ret = libusb_claim_interface(handle, 2); after the device was opened successfully) to make sure libusb itself to worked out, but still I got the same err.

so i think if this was the error of the libusb itself.

platform:
windows 10 64bits
with Winusb installed.

mcuee, 2018-06-14:
You need to provide more details about your device -- use USBView to post the full descriptors of the device.

mcuee, 2018-06-14:
And this is a USB composite device, how do you install the driver? Try to use Zadig to install the WinUSB to the USB composite parent to see if that helps.

guanguanboy, 2018-06-15:
@mcuee  thanks for your replay. below is the full descriptors:

[Port1]  :  USB Composite Device

Is Port User Connectable:         no
Is Port Debug Capable:            no
Companion Port Number:            0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "Astra Pro HD Camera"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: High (is not SuperSpeed or higher capable)
Device Address:                    0x06
Open Pipes:                           1

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x2BC5 = Orbbec 3D Tech. Int'l Inc.
idProduct:                       0x0501
bcdDevice:                       0x2702
iManufacturer:                     0x02
     English (United States)  "Astra Pro HD Camera"
iProduct:                          0x01
     English (United States)  "Astra Pro HD Camera"
iSerialNumber:                     0x00
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0010 = 1 transactions per microframe, 0x10 max bytes
bInterval:                         0x06

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x01D1  -> Validated
bNumInterfaces:                    0x02
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0x80  -> Bus Powered
MaxPower:                          0xFA = 500 mA

          ===>IAD Descriptor<===
bLength:                           0x08
bDescriptorType:                   0x0B
bFirstInterface:                   0x00
bInterfaceCount:                   0x02
bFunctionClass:                    0x0E  -> Video Interface Class
bFunctionSubClass:                 0x03  -> Video Interface Collection
[... 428 lines cut ...]
wMaxPacketSize:                  0x1320 = 3 transactions per microframe, 0x320 max bytes
bInterval:                         0x01

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x01
bAlternateSetting:                 0x06
bNumEndpoints:                     0x01
bInterfaceClass:                   0x0E  -> Video Interface Class
bInterfaceSubClass:                0x02  -> Video Streaming Interface SubClass
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x05  -> Isochronous Transfer Type, Synchronization Type = Asynchronous, Usage Type = Data Endpoint
wMaxPacketSize:                  0x1400 = 3 transactions per microframe, 0x400 max bytes
bInterval:                         0x01

          ===>Additional Error Checking<===
PASS: number of uncompressed-frame frame descriptors (3) == number of frame descriptors (3) specified in uncompressed format descriptor(s)
PASS: number of MJPEG frame descriptors (3) == number of frame descriptors (3) specified in MJPEG format descriptor(s)

guanguanboy, 2018-06-15:
BTW, I already used Zadig and installed the WinUSB to the USB composite parent, but it still doesn't work.
what i said I installed the driver,  what relly mean is that I installed the WinUSB to the USB composite parent using Zadig.

mcuee, 2018-06-15:
When you use the following, you assume it has Interface 2, but it only has Interface 0 and Interface 1 (group together using IAD). Interface 1 has multiple alternative settings. You may want to post more details of what you are trying to do and what is the problem and post the debug logs.
```ret = libusb_claim_interface(handle, 2)

mcuee, 2018-06-15:
BTW, I am not so sure libuvc example.c will work for your device without changes, I do not know the details about libuvc, you may want to ask in libuvc side first on how to use it.

mcuee, 2018-06-15:
I just take a quite look at libuvc, it needs isoc transfer support, currently libusb's WinUSB backend does not support isoc transfer mode.

You can try to use libusbK driver and the latest version of libusb 1.0.22 adds isoc transfer support for libusbk. Please try agin with libusbK.

mcuee, 2018-06-15:
This is what I get for Microsoft LifeCam VX-800, using libusb 1.0.22 and libusbk driver.
```
ephwy@ephwy-HP MINGW64 /c/work/libuvc/libuvc/build
$ ./example.exe
UVC initialized
Device found
Device opened
Streaming...
Done streaming.
Device closed
UVC exited
unsupported descriptor subtype VS_STILL_IMAGE_FRAME
unsupported descriptor subtype VS_COLORFORMAT
DEVICE CONFIGURATION (045e:0766/[none]) ---
Status: idle
VideoControl:
        bcdUVC: 0x0100
VideoStreaming(1):
        bEndpointAddress: 130
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 16
                  GUID: 5955593200001000800000aa00389b71 (YUY2)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 00
                          size: 640x480
                          bit rate: 196608000-196608000
                          max frame size: 614400
                          default interval: 1/30
                          interval[0]: 1/30
                        FrameDescriptor(2)
                          capabilities: 00
                          size: 352x288
                          bit rate: 196608000-196608000
                          max frame size: 202752
                          default interval: 1/30
                          interval[0]: 1/30
                        FrameDescriptor(3)
                          capabilities: 00
                          size: 320x240
                          bit rate: 196608000-196608000
                          max frame size: 153600
                          default interval: 1/30
                          interval[0]: 1/30
                        FrameDescriptor(4)
                          capabilities: 00
                          size: 176x144
                          bit rate: 196608000-196608000
                          max frame size: 50688
                          default interval: 1/30
                          interval[0]: 1/30
                        FrameDescriptor(5)
                          capabilities: 00
                          size: 160x120
                          bit rate: 196608000-196608000
                          max frame size: 38400
                          default interval: 1/30
                          interval[0]: 1/30
END DEVICE CONFIGURATION
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 3072
bInterfaceNumber: 1
attempt to claim already-claimed interface 1
```

mcuee, 2018-06-15:
Using another webcam (Prolink PCC3220) and it seems to work with libusbK driver.
```
C:\work\libuvc\libuvc\build\bin>example.exe
UVC initialized
Device found
unsupported descriptor subtype VS_STILL_IMAGE_FRAME
unsupported descriptor subtype VS_COLORFORMAT
Device opened
DEVICE CONFIGURATION (0ac8:3420/[none]) ---
Status: idle
VideoControl:
        bcdUVC: 0x0100
VideoStreaming(1):
        bEndpointAddress: 130
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 16
                  GUID: 5955593200001000800000aa00389b71 (YUY2)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 00
                          size: 640x480
                          bit rate: 768000-196608000
                          max frame size: 614400
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/14
                        FrameDescriptor(2)
                          capabilities: 00
                          size: 352x288
                          bit rate: 768000-196608000
                          max frame size: 202752
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/14
                        FrameDescriptor(3)
                          capabilities: 00
                          size: 320x240
                          bit rate: 768000-196608000
                          max frame size: 153600
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/14
                        FrameDescriptor(4)
                          capabilities: 00
                          size: 176x144
                          bit rate: 768000-196608000
                          max frame size: 50688
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/14
                        FrameDescriptor(5)
                          capabilities: 00
                          size: 160x120
                          bit rate: 768000-196608000
                          max frame size: 38400
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/14
END DEVICE CONFIGURATION
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 2688
bInterfaceNumber: 1
attempt to claim already-claimed interface 1
Streaming...
Done streaming.
Device closed
UVC exited
```

mcuee, 2018-06-15:
It seems to me that you need to modify the example to be suitable for different webcams. I do not think this is a libusb issue. You may want to ask in libuvc side.

guanguanboy, 2018-06-15:
@mcuee  
thanks for your reply.
I am trying to using libusbK. I have a question, how did you compile libusb 1.0.22, did you using minGW or just download the compiled binary from website?

mcuee, 2018-06-15:
In this case, I build libuvc example under MSYS2 which has libusb-1.0.22 binary package. But you should be able to use the binary package from libusb website as well.

mcuee, 2018-06-20:
Closing this. 
To OP: please use libusb mailing list for further support.

closed 2018-06-20 by mcuee

jiang-bin-abel, 2022-01-18:
I also found the same problem when there are multiple alternates under an interface.