libusb/libusb #447: libusb_open with -5 return

UnisNick, 2018-06-19:
I tried to use libusb_get_string_descriptor_ascii to get serial number from a CCID smartcard reader, but before that I need to allocate a handle for the device. Either I chose libusb_open_device_with_vid_pid or libusb_open, -5 returned, which is "LIBUSB_ERROR_NOT_FOUND". Same procedure worked on my USB mouse and keyboard. Please can anyone advise? Many many thanks.

mcuee, 2018-06-19:
Please use libusb mailing list for technical support like this. Thanks.

Anyway, are you using Windows? If yes, did you install a supported driver
like WinUSB?
-- 
Xiaofan

mcuee, 2018-06-28:
Close for now. Please use libusb mailing list for further support.

closed 2018-06-28 by mcuee

TerryChan, 2018-07-04:
I got the same issue on windows10(libusb_open failed with libusb_error_not_found), and I installed the libusb driver(libusb0.sys, libusb0.dll), do you know how to fix it? do I need to install WinUSB still?

toBeGeekNo1, 2020-10-14:
I have the same problem on win7. Is there any way to solve it？