libusb/libusb #449: cannot install if target path contains blank

AndreasFMueller, 2018-06-29:
libusb-1.0.22 cannot be installed if the prefix contains a blank. The following error message is displayed:

libtool: install: ranlib /usr/local/lib/libusb-1.0.a
/bin/sh: /Volumes/Macintosh: No such file or directory

On this particular system, /usr/local/lib is a symbolic link to /Volumes/Macintosh HD 2/local/lib
The symbolic path does not have any blanks, but the physical target path has blanks, causing installation to fail.

This probably is an issue with libtool.

hjelmn, 2018-07-09:
Will take a look.

dickens, 2020-01-14:
Sorry, this is a hard limitation of autoconf.  See the "special characters" section in the [autoconf documentation](https://www.gnu.org/software/autoconf/manual/autoconf-2.69/html_node/Installation-Directory-Variables.html)

closed 2020-01-14 by dickens