libusb/libusb #455: libusb_reset_device not working on macOS

gniezen, 2018-07-26:
It seems that `libusb_reset_device` has been broken on macOS [since El Capitan](https://sourceforge.net/p/libusb/mailman/message/34744234/). I discovered this in a [node-usb thread](https://github.com/tessel/node-usb/issues/159) that mentions:

> Note that OSX >= 10.11 silently ignores reset commands from libusb

The reason why this is an issue is because some chips, like the TI TUSB3410, needs to be soft reset and re-enumerated after a firmware update. The USB cable (with TUSB3410 chip) I'm targeting requires the firmware to be loaded in every time it is connected, and after the firmware is loaded it will not respond to anything until the soft reset. It works fine on Linux, but it took me much too long to figure out that soft reset just fails silently on macOS.

Applying the change in the original libusb thread above, that is, changing 
```
kresult = (*(dpriv->device))->ResetDevice (dpriv->device);
```
to 
```
kresult = (*(dpriv->device))->USBDeviceReEnumerate (dpriv->device, 0);
```
works great. I guess my question is, if it is known that `ResetDevice` fails silently on macOS, and Apple were [apparently the ones to suggest](https://sourceforge.net/p/libusb/mailman/message/34744469/) using `usbDeviceReEnumerate` instead, why hasn't this been changed since 2016? If there is a good reason to keep using `ResetDevice`, can we then either have it always re-enumerate (not only if one of the descriptors have changed), or have a separate function for re-enumerating devices?

Based on [Apple's description](https://developer.apple.com/documentation/iokit/iousbdeviceinterface187/1559027-usbdevicereenumerate?language=objc) of `usbDeviceReEnumerate`, it does essentially reset the device:

> This function will send a terminate message to all clients of the IOUSBDevice (such as IOUSBInterfaces and their drivers, as well as the current User Client), emulating an unplug of the device. The IOUSBFamily will then enumerate the device as if it had just been plugged in. This call should be used by clients wishing to take advantage of the Device Firmware Update Class specification. The device must be open to use this function.

gniezen, 2018-07-26:
@LudovicRousseau You may notice that I linked to that thread in the first sentence already. :wink:

hjelmn, 2018-07-26:
My bad. I was going to make this change then got distracted by something else.

I can make this change now.

hjelmn, 2018-07-26:
FYI: Looking at the last available OSS version of IOUSBFamily (10.8.5) it looks like USBDeviceReEnumerate effectively resets the device.

hjelmn, 2018-07-26:
#456 should fix this.

closed 2018-07-26 by hjelmn

gniezen, 2018-07-26:
Wow, that was super-quick, thanks @hjelmn ! :smile:

yplaksyuk, 2018-10-16:
Unfortunately, this commit breaks connection with LimeSDR Mini. It just doesn't work at all. Rolling back these changes restores normal operation.

juribeparada, 2019-03-10:
I can also report similar problem with https://github.com/libusb/libusb/pull/456, it breaks LimeSDR-Mini SDR board (which uses FT601) on macOS. Reverting only this change for the latest libusb code, will fix the issue.

hjelmn, 2019-03-10:
The problem is this fix is necessary to actually reset the device. The old function is essentially a no-op. I don't know what will need to be done to deal with the issue on the LimeSDR side.

juribeparada, 2019-03-10:
OK, I understand, then the solution should be on the driver side. Probably LimeSDR driver will need an update for the next libusb release.

hjelmn, 2019-04-04:
@juribeparada I found an issue with the fix. It isn't correctly returning ```LIBUSB_ERROR_NOT_FOUND``` when the libusb device structure is no longer valid. I am working on a fix and should have it pushed today.

hjelmn, 2019-04-04:
@juribeparada Can you try https://github.com/hjelmn/libusb/commit/b9770b212d7901843316ede9e0243a110bdff15d

and see if it resolves the issue for you?

juribeparada, 2019-04-04:
Thank you @hjelmn to work on this issue. I will test soon and inform you.

juribeparada, 2019-04-04:
OK, I tested your patch and definitely solves my issue. LimeSDR-Mini is working very well now. Thank you very much!

hjelmn, 2019-04-04:
@juribeparada Thanks for testing that. Will run some additional tests and push it to master. Will start the release process soon.

hjelmn, 2019-04-07:
@juribeparada The fix is included in 1.0.23-rc1.

juribeparada, 2019-04-07:
Thank you very much

commit c14ab5fc (2018-07-26): os/darwin: switch from using ResetDevice to USBDeviceReEnumerate

In newer versions of macOS the ResetDevice function does nothing. For
these systems we have to use USBDeviceReEnumerate. This should also
work for older versions os MacOS X.

Fixes #455