libusb/libusb #458: windows: 1/10 possible open device failure or claim interface failure

nxpfrankli, 2018-07-29:
I poll all devices every 100ms.  If found new device attached and it is what I interested, I try to open it and send some data.  My attached devices is HID devices. 

My code is in https://github.com/codeauroraforum/mfgtools/blob/master/libuuu/usbhotplug.cpp

There are NOT problem in linux platform.

diabolo38, 2018-07-31:
I only know few case where open fail.
You try to open a device already open either in our app or some other app.

You could discover and think of two device but they are same. 

If you device disconnect and reconnect (got soft reset also) on same port it may cos you open ssue if you had it open. This hapen always with USB UART bridge and their no way to recover until you close the open device (terminal app) eject and inset device again. (Same on Linux)

I some rare case init/fini context mishandling can cos failure i ll sugest use of multi context and not mess up taking
Dev ref but only use one open dev per Context (this is the only way to avoid long trerm leak)

rarely a device appear on bus but is not fully ready and nak few control (read desc ...) Causing open failurebor bad device enum.

When device is composite some interface may appear while win  full device driver powr up is not completely over ops at that time may fail hapen to me very rarely with ndis +MSC + customs gadget on re enum leak stress test

nxpfrankli, 2018-07-31:
Thanks,  My same device attached to the same port. 

in winusb_get_device_list
					dev = usbi_alloc_device(ctx, session_id);

printf("New device create %p %d pass %d\n", dev, session_id, pass);

I found sometime 
    New device create 0000018655B10F50 117 pass 3

But sometime follow HID have not set for session_id. 
Look like HID have not install for this device after device scaned at pass 3.

If I delay 200ms and call get_list again, look like workaround this problem

mcuee, 2018-08-01:
Thanks for the report. I think this is not an libusb issue but rather Windows OS limitation. So closing this for now. 

BTW, FYI, libusb Windows does not support hotplug as of now.

closed 2018-08-01 by mcuee