libusb/libusb #461: set_composite_interface function fails to get the interface number under Windows

joska86, 2018-08-08:
I am trying to connect to my Huawei mobile phone via USB and communicate with it as ADB (Android Debug Bridge) does. But it seems, something went wrong, because in the file **windows_winusb.c** the function set_composite_interface failed to get the valid interface number value (I debugged it, so the problem is there for sure), because the _device_id_ parameter does not contain any subsrting prefixed with "MI_" which should contain the interface number.

The piece of the debug log:

`[ 0.199000] [00000e00] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.208000] [00000e00] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.215000] [00000e00] libusb: debug [winusb_get_device_list] setting composite interface for [1BBC]:
[ 0.225000] [00000e00] libusb: warning [set_composite_interface] failure to read interface number for USB\VID_12D1&SUBCLASS_42&PROT_01\8&277680DE&0&0004, using default value
[ 0.231000] [00000e00] libusb: debug [set_composite_interface] interface[0] = \\.\USB#VID_12D1&SUBCLASS_42&PROT_01#8&277680DE&0&0004#{F72FE0D4-CBCB-407D-8814-9ED673D0DD6B}`

[image]

[image]

As you can see, the Device Instance Path which is used in libusb code too:
`			if (!pSetupDiGetDeviceInstanceIdA(*dev_info, &dev_info_data, dev_id, sizeof(dev_id), NULL)) {
				usbi_warn(ctx, "could not read the device instance ID for devInst %X, skipping",
					  dev_info_data.DevInst);
				continue;
			}`
did not returned any MI_ prefixed string.

mcuee, 2018-08-10:
Please provide more details.
1) post the usb descriptor of your device
2) provide the details how you install the driver for your device.
3) provide the code and debug log

joska86, 2018-08-13:
1.
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x12D1	idVendor
0x107D	idProduct
0x0318	bcdDevice
0x01	iManufacturer   "HUAWEI"
0x02	iProduct   "HUAWEI MLA-L11"
0x03	iSerialNumber   "GLDDU16A30003604"
0x01	bNumConfigurations

The interface (0x03) which I want to use to communicate to the phone is:
0x09	bLength
0x04	bDescriptorType
0x03	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x42	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x09	iInterface   "ADB Interface"

2

Just plug in and Windows installed the driver, but I've tried to install manually the driver (HiSuite Huawei package)

3
The code snippet:

```
libusb_device_handle* handle_raw = nullptr;
	rc = libusb_open(device, &handle_raw);
	unique_device_handle handle(handle_raw);
	std::string deviceSerial;
	if (rc == 0) {
		
		std::cout << "successfully opened adb device at " << deviceAddress << std::endl;

		deviceSerial.resize(255);
		rc = libusb_get_string_descriptor_ascii(handle_raw, device_desc.iSerialNumber,
			reinterpret_cast<unsigned char*>(&deviceSerial[0]),
			deviceSerial.length());
		if (rc == 0) {
			std::cout << "received empty serial from device at " << deviceAddress<<std::endl;
			return;
		}
		else if (rc < 0) {
			std::cout << "failed to get serial from device at " << deviceAddress
				<< libusb_error_name(rc) <<std::endl;
			return;
		}
		deviceSerial.resize(rc);
		rc = libusb_claim_interface(handle.get(), interface_num);
		if (rc != 0) {
			std::cout << "failed to claim adb interface for device '" << deviceSerial << "'"
				<< libusb_error_name(rc) << std::endl;
			return;
		}

		rc = libusb_set_interface_alt_setting(handle.get(), interface_num, 0);
		if (rc != 0) {
			std::cout << "failed to set interface alt setting for device '" << deviceSerial
				<< "'" << libusb_error_name(rc) << std::endl;
			return;
		}
[... 120 lines cut ...]
[ 0.286000] [00000914] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.288000] [00000914] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.289000] [00000914] libusb: debug [handle_events] poll() 2 fds with timeout in 984ms
[ 0.291000] [00000914] libusb: debug [handle_events] poll() returned 1
[ 0.293000] [00000914] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
[ 0.294000] [00000914] libusb: debug [usbi_remove_pollfd] remove fd 2
[ 0.296000] [00000914] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 34
[ 0.298000] [00000914] libusb: debug [libusb_release_interface] interface 0
[ 0.300000] [00000914] libusb: debug [auto_release] auto-released interface 0
[ 0.301000] [00000914] libusb: debug [usbi_handle_transfer_completion] transfer 0087B478 has callback 00B12D10
[ 0.303000] [00000914] libusb: debug [sync_transfer_cb] actual_length=34
[ 0.305000] [00000914] libusb: debug [libusb_free_transfer] transfer 0087B478
[ 0.306000] [00000914] libusb: debug [libusb_claim_interface] interface 3
[ 0.308000] [00000914] libusb: debug [composite_claim_interface] unsupported API call for 'claim_interface' (unrecognized device driver)
[ 0.310000] [00000914] libusb: debug [libusb_close]
[ 0.312000] [00000914] libusb: debug [libusb_get_device_descriptor]
[ 0.314000] [00000914] libusb: debug [libusb_get_device_descriptor]
[ 0.315000] [00000914] libusb: debug [libusb_get_device_descriptor]
[ 0.317000] [00000914] libusb: debug [libusb_get_device_descriptor]
[ 0.319000] [00000914] libusb: debug [libusb_get_device_descriptor]`

Problematic line:
**[ 0.202000] [00000914] libusb: warning [set_composite_interface] failure to read interface number for USB\VID_12D1&SUBCLASS_42&PROT_01\8&309DA40&0&0003, using default value**

There is no MI_prefixed string to determine the interface number (however I assume that the last element in the string (0003) is the real interface number)

dickens, 2020-01-13:
This device must be using usbccgp as its driver, which is strange if it is not a composite device.  What is this device's parent?  Can you show what the Device Manager looks like when viewing by connection?

dickens, 2020-08-17:
Closing since there has been no response to the request for information in over 6 months.  Please reopen if there is additional information to provide.

closed 2020-08-17 by dickens