libusb/libusb #463: libusb_get_device_list returns zero in Cocoa app

kianm, 2018-08-11:
https://stackoverflow.com/questions/51785153/libusb-returns-zero-devices-on-cocoa-app

voxels, 2018-09-11:
+1 this issue 

I made two test XCode projects to demonstrate this issue:
https://github.com/voxels/TestLibUsb
https://github.com/voxels/TestLibUsbCocoa

I found this issue testing an Intel RealSense D400.
The test code [enumerates](https://github.com/voxels/TestLibUsb/blob/master/TestLibUsb/main.cpp#L14-L41) the device list.

By using LIBUSB_DEBUG=4, I was able to find the following console output:

**TestLibUsb** (the successful command line tool)
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000013] [00000307] libusb: debug [libusb_init] created default context
[ 0.000064] [00000307] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000284] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x7cbc4813
[ 0.000411] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x7cbc4813
[ 0.007035] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.007049] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.007052] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.007055] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xef
[ 0.007057] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x02
[ 0.007059] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.007061] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.007063] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.007065] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8511
[ 0.007067] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x7252
[ 0.007069] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.007071] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.007073] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x03
[ 0.007075] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.007100] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.007103] [00000307] libusb: debug [process_new_device] allocating new device in context 0x100749120 for with session 0x7cbc4813
[ 0.007109] [00000307] libusb: debug [process_new_device] found device with address 3 port = 5 parent = 0x0 at 0x100749c8a
[ 0.007172] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x13c5680f3
[ 0.007426] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID 0x13c5680f3 against cached device with sessionID 0x7cbc4813
[ 0.007432] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x13c5680f3
[ 0.007753] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.007762] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.007777] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0201
[ 0.007782] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xef
[ 0.007784] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x02
[ 0.007786] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.007788] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.007790] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.007792] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8296
[ 0.007794] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0070
[ 0.007796] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.007798] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.007800] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.007967] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.007999] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.008002] [00000307] libusb: debug [process_new_device] allocating new device in context 0x100749120 for with session 0x13c5680f3
[ 0.008006] [00000307] libusb: debug [process_new_device] found device with address 5 port = 6 parent = 0x0 at 0x10044b62a
[ 0.008073] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x90d7d0ec92a
[ 0.008186] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID 0x90d7d0ec92a against cached device with sessionID 0x13c5680f3
[ 0.008189] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID 0x90d7d0ec92a against cached device with sessionID 0x7cbc4813
[ 0.008191] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x90d7d0ec92a
[ 0.008485] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.008494] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[... cut ...] g preferences outside an application's container requires user-preference-read or file-read-data sandbox access, detaching from cfprefsd
```

The final message `[User Defaults] Couldn't read` seems to be "spurious" according to SO.  

I don't know enough about this to make a reasoned guess.  The suggested path seems to be using IOKit instead of libUSB, however, it would be nice to avoid rewriting the librealsense dependency if possible.

Any thoughts about how to proceed are greatly appreciated!

hjelmn, 2018-09-11:
I will take a look now. Having successfully used libusb in Cocoa apps this is a little surprising.