libusb/libusb #466: Given a `libusb_device`, is there a way to get the already opened `libusb_device_handle`?

rpici, 2018-08-17:
The libusb library already has a function to get the `libusb_device` associated with a `libusb_device_handle`:
>`libusb_device* libusb_get_device( libusb_device_handle *dev_handle );`

...but I can't find the opposite function to get the `libusb_device_handle` associated with a `libusb_device`, if such a `libusb_device_handle` exists:
>`libusb_device_handle* libusb_get_existing_device_handle_or_null_if_none( libusb_device *device );`

Note that I do not want to open a new `libusb_device_handle` using `libusb_open(...)`; I want to get any existing `libusb_device_handle` on a `libusb_device`, or `NULL` if no such already-opened `libusb_device_handle` exists.

Does such a function exist?

If not, is there another way to get the existing (i.e. already opened) `libusb_device_handle`, if any, from a `libusb_device`?

The reason I ask is due to the following hotplug example:
http://libusb.sourceforge.net/api-1.0/hotplug.html

If I modify the hotplug example (to handle more than 2 hotplug events, and to output the vid/pid of the handle that gets closed in the callback) and run it, and I have more than one USB device, and I plug them in and unplug them in this order:
>plug in device A
>plug in device B
>unplug device A

...then when device A is unplugged, `libusb_close(handle);` closes the handle for device B, not device A.  This is because when device B was plugged in, in `hotplug_callback()` the static `handle` variable was set for device B.  Then when A is unplugged, `handle` is still pointing to the `libusb_device_handle` for device B from the previous invocation of the callback.

To fix that, I think would need to find any already open `libusb_device_handle` associated with a `libusb_device`, and call `libusb_close` on that found `libusb_device_handle`.

So if I have a `libusb_device`, is there a way to obtain any already-opened `libusb_device_handle` associated with that `libusb_device`?

Am I asking the wrong question (XY problem)?

diabolo38, 2018-08-18:
If you have open  the device then you should have the handle, the lib can likely do what you ask but to me you should handl it yourself. 
Also it is not forbidden to have multiple handle on the same device I use this feature heavily

rpici, 2018-08-20:
Replying to @diabolo38, commented 2 days ago
> 
> If you have open the device then you should have the handle, the lib can likely do what you ask but to me you should handl it yourself.
> Also it is not forbidden to have multiple handle on the same device I use this feature heavily

Ok.

If a map of
>`libusb_device*` -> list of already-open `libusb_device_handle*`'s

isn't provided by the library, then I can maintain such a map myself, per your suggestion.  I just wanted to know if such a mapping is already provided by the library, but apparently it is not...?

diabolo38, 2018-08-20:
I may be wrong but at least under window the lib do not manage a shared memory between using process for shared data that is what you kind of ask. 
Each process has it own Copy of the lib global data.

Under Linux Macos or window (but libusbk back end) it is not possible to open a device multipletime in different  process so if device is already open it is necessarly in  your procress and you must know about it if your writing the app.   if you try open plus claim and it fail it is open I another process or you have no perm and  it is what expected.
The only case I see you can t manage it is your writing a DLL that use libusb in some app  already using libusb or linked with a dll or lib also using libusb and is also using  the same device/interface. What is a problem. Open claim may sucees but result of non controled multiple device use will be a disaster.

dickens, 2018-08-26:
It may be possible to open a device multiple devices, thus there is no one-to-one mapping and such a function doesn’t make sense.  I’m closing this because this isn’t really an issue.

closed 2018-08-26 by dickens

rpici, 2018-08-27:
@dickens If a function that assumes a one-to-one mapping doesn't make sense, then how about a function that assumes a one-to-many mapping?  e. g.:

>`libusb_device_handle** libusb_get_existing_device_handles_or_null_if_none( libusb_device *device );`

or in C++:

> `std::vector<libusb_device_handle*> libusb_get_existing_device_handles_or_empty_vector_if_none( libusb_device *device );`

That's what I'm doing now, outside the library, to get the example I described above to work correctly.