libusb/libusb #468: libusb_set_configuration 2 Fails with error LIBUSB_ERROR_NOT_FOUND

KORuL, 2018-08-30:
Good afternoon! I'm trying to translate the device to libusb_set_configuration in configuration number 2, but I get a -5 or LIBUSB_ERROR_NOT_FOUND error in response. What am I doing wrong?

```
static int setConfiguration(libusb_device *dev)
{
	int level = 0;
	struct libusb_device_descriptor desc;
	libusb_device_handle *handle = NULL;
	char description[256];
	char string[256];
	int ret;
	uint8_t i;

	ret = libusb_get_device_descriptor(dev, &desc);
	if (ret < 0) {
		fprintf(stderr, "failed to get device descriptor");
		return -1;
	}
	ret = libusb_open(dev, &handle);
	ret = libusb_reset_device(handle);
	if(ret == LIBUSB_SUCCESS) {
		printf("reset success\n");

		ret = libusb_set_configuration(handle, 2);
		
		if(ret == LIBUSB_SUCCESS) {
			printf("set configuration success\n");
		}
		else
			printf("set configuration fail %d\n", ret);
	}
	else
		printf("reset fail %d\n", ret);

	if (handle)
		libusb_close(handle);

	return 0;
}

int main(int argc, char *argv[])
{
	libusb_device **devs;
	ssize_t cnt;
	int r, i;

	if (argc > 1 && !strcmp(argv[1], "-v"))
		verbose = 1;

	r = libusb_init(NULL);
	if (r < 0)
		return r;

	cnt = libusb_get_device_list(NULL, &devs);
	if (cnt < 0)
		return (int)cnt;

	for (i = 0; devs[i]; ++i) {
		
		if(i == 1)
			setConfiguration(devs[i]);

	}

	libusb_free_device_list(devs, 1);

	libusb_exit(NULL);
	return 0;
}

mcuee, 2018-09-12:
Does the device have Configuration 2? Very few device have Configuration 2. And what is the OS and driver used? WinUSB does not support non-default configuration.

dickens, 2018-09-14:
What operating system are you trying to do this on?  Please post the device's descriptors (device & configuration).

mcuee, 2019-01-17:
Close since the OP does not come back.

closed 2019-01-17 by mcuee