libusb/libusb #469: Is sudo required for retrieving device string descriptors?

shinobisoft, 2018-08-31:
So I downloaded, zip version, built and installed libusb as a shared library ( I assume ). From there I built one of the examples, testlibusb.c, and then ran the example. On first run, I noticed no device names being printed to stdout. Added a couple of printf() statements in the example code and rebuilt. After running the example again, still no device strings being printed and my printf() statements weren't getting output either. So... I started to surmise the I may need root permissions. I executed my example program via sudo and lo and behold, device names appeared in the output. From there, I checked the return value of libusb_get_string_descriptor_ascii() and it's returning LIBUSB_ERROR_ACCESS.

I've used usb4java in the past which is a java wrapper for libusb 1.0 and root permissions are not required to retrieve device names. Do I need to add a custom file to /etc/rules.d to allow libusb to work properly without using sudo?

shinobisoft, 2018-09-12:
@mcuee The title may come across as a question but this is not a question. It's an issue. I've built this library twice, using y'all's bootstrap.sh and configure, and end up with the same end result... No device strings from built-in or attached hardware, without having to call sudo first...

diabolo38, 2018-09-13:
If you add yourself to plugdev group or add add a rules for your device then sudo will not be required

shinobisoft, 2018-09-14:
Well this is strange... The output of 'id' for my 18.04 box...
`id
uid=1000(shinobisoft) gid=1000(shinobisoft) groups=1000(shinobisoft),4(adm),24(cdrom),27(sudo),30(dip),33(www-data),46(plugdev),108(lpadmin),124(sambashare)`

dickens, 2018-09-14:
The kernel by default gives USB device nodes 0600 permissions, so only root can access them.  You need to set up udev/systemd rules for the device nodes of interest, or run as root.

closed 2018-09-14 by dickens