libusb/libusb #470: "libusb_bulk_transfer" return -1

troies, 2018-09-03:
I am using QT5.7.1+libusb to work with a development board, and I am aimed to use libusb to do the uboot image transfer, but I get an error -1 when I call the libusb_bulk_transfer function, here is the flow of my code:

handle = libusb_open_device_with_vid_pid(NULL, TARGET_VID, TARGET_PID);
ret = libusb_kernel_driver_active(handle, 0); // here the ret = 0;
ret = libusb_claim_interface(handle, 0);
unsigned char data[512] = {0x01, 0x02, 0x03, 0x04};
int actual_length = 0;
ret = libusb_bulk_transfer(handle, 129, data, 512, &actual_length, 0);

But the final ret = -1, could any one figure out what is wrong with my code?

mcuee, 2018-09-12:
Please use libusb mailing list for technical support. This github is for reporting of libusb issues.

closed 2018-09-12 by mcuee

troies, 2018-09-12:
well, thanks!

793128483, 2018-11-27:
想问，一次传输512个字节能成功吗，我想问能否用两个端点传输每个端点传输128个字节，