libusb/libusb #478: Crash in libusb_bulk_transfer when device is removed right after it was discovered. (OSX)

unier, 2018-09-23:
When discovering an usb device with bulk transfer capability, and removing it very fast after it was discovered results in crash in libusb_bulk_transfer call. 

Callstack (EXC_BAD_ACCESS):
list_del
remove_from_flying_list
libusb_submit_transfer
do_sync_bulk_transfer
libusb_bulk_transfer

This might have to do with thread safety the call is from a different thread than the one the device was discovered in, but I am unsure what should be synchronised to prevent this crash from happening. It is not reproducible every time I try it, but just every 4th time or so. So that makes me think that I am using this in a non-thread safe way.

This typical happens when I have an usb device connected that requires more power inrush than 500mA that can be given. Leads to a discovery and then the usb device is shutdown when powered up.

dickens, 2020-01-13:
Is there some thread that is closing the device while another thread sits waiting in `libusb_bulk_transfer()`?

seanm, 2020-03-20:
Building with Thread Sanitizer might reveal any threading problem...

mcuee, 2020-12-07:
@unier Do you still encounter this issue?

mcuee, 2021-02-26:
Close this for now due to inactivity. Please reopen if more info is provided.

closed 2021-02-26 by mcuee