libusb/libusb #481: Failed to open Kinect V2

sdewailly, 2018-09-27:
Hello,

I am trying to use a kinect V2 via usb 3 port. In fact, after lauching an app that is based on libusb library an error appeared!
It seems this app doesn't have enough permissions to access the device because another process have exclusive access. I don't know more about the architecture nor the implementation of libusb software...
I was sudoing as well as setting the variable environment LIBUSB_DEBUG to 3 as it was written.
No good result!
[libusb.txt](https://github.com/libusb/libusb/files/2425832/libusb.txt)

Thanks in advance, Sébastien

PS. I'm using a mac

diabolo38, 2018-09-28:
Likely it is composite device and the interface you try claiming is in use by system (std class device like audio,keyboard....). 
They are api to detach kernel driver that may be required used here.

hjelmn, 2018-10-01:
More than likely google will point the user at how to use kinect v2 on macOS.

mcuee, 2020-07-29:
Closed for now as this is not a libusb issue.

closed 2020-07-29 by mcuee