libusb/libusb #484: Can't send control requests to non zero interface (Windows)

garfa, 2018-10-12:
If I send a control request with recipient == interface and index != 0 the index sent in the setup packet to the device is always zero. 

I observed this with Winusb and Libusbx backends 

I resolved this for my usage in the Winusb case at least by adding the patch below to winusbx_submit_control_transfer() 

```current_interface = get_valid_interface(transfer->dev_handle, USB_API_WINUSBX);
	if (current_interface < 0) {
		if (auto_claim(transfer, &current_interface, USB_API_WINUSBX) != LIBUSB_SUCCESS)
			return LIBUSB_ERROR_NOT_FOUND;
	}
  // ------------------------ start of patch -----------------------------------------
      if (setup->RequestType & LIBUSB_RECIPIENT_INTERFACE
        && setup->Index)
      {
        current_interface = setup->Index;
      }
  // ------------------------ end of patch -----------------------------------------

	usbi_dbg("will use interface %d", current_interface);
```
The details of why this is necessary are in the remarks at 

https://docs.microsoft.com/en-us/windows/desktop/api/winusb/nf-winusb-winusb_controltransfer

ohgeek, 2018-10-15:
Thanks, I have just run into this problem. :)

dickens, 2020-01-13:
This should be fixed with #398 and #567.  OP can you test against master and report back?

mcuee, 2020-08-21:
@garfa Please check. Thanks.

mcuee, 2021-01-23:
Close this for now. Please reopen if the issue is still there.

closed 2021-01-23 by mcuee