libusb/libusb #485: less data length in async bulk transfer with 2 cameras

zheng-rong, 2018-10-19:
Hi all,

Firstly sorry for submitting this ask-for-support post here. I know sourceforge mailling list is the right place, but sending email failed with message indicating my IP is blocked, though I tried different Wifi.

I am developing an application using libusb with a custom USB camera (using CY7C68013 USB chip).
I am using USB2.0 bulk (512Byte) asynchronous transfer mode.

On the device side, the image frame size = 360960=512*705, so in the firmware the device will send 705 packets ended with a zero-length packet (totally 706 pkts). The camera streams images frame at about 30fps.

On the host side, the application code works in async mode. It initialize a transfer queue (e.g. 16) and submit the transfer request. Each transfer will request 706 packets with 5 seconds timeout setting. In the xfer_callback, it will check the data length and re-submit the transfer. I run the libusb_handle_events in main loop. ( also tried in an independent thread)

When running one camera, the application code works fluently without any problem.

But when running 2 cameras in the same computer from another terminal (using the same executable binary file, but will request different camera serial number), these two cameras will lost data. Some transfers are completed, but the actual transfer length is less than the expected length(e.g. 360852 < 360960). It seems that the 2nd camera interfere the data transfer, which results in the incomplete data frame of 2 cameras.

Also I tried to implement the 2 cameras instances in my code. In the code I submit transfers in (A-B-A-B-A-B-A-B....) mode, and use one xfer_callback function. In the callback function I use the different  device_handle to identify the different camera device. libusb_handle_event runs in a separate thread. But in the result I got the same problem.

Another try is that, I defined the camera device class and declare 2 camera objects. For each camera, I start a new thread to configure, submit transfer and run libusb_handle_events. But got the same problem.

1. What may be the reason of this problem?

-   Bandwidth? I don't think so. I tested off-the-shelf USB2.0 camera on the same PC, 2 cameras can works fluently at 60fps for each. (the camera interface is USB2.0, and PC interface is USB3.0).
-   Multi-thread?
-   Any other possible reasons?

2. By the way, I noticed the "context" concept in libusb, but I can't understand it clearly. When should I use different context rather than default context?

- If I define a camera class and declare 2 camera instance in my app code, is it necessary to use two different context?
- If only one camera is used in app code, but 2 app are executed in different linux terminal, in this case libusb works in different process, so the default context can be used (i.e. using libusb_init(NULL)), right?
-  In my current code implementation, I am using libusb_init(NULL) i.e. default context for the USB transfer. Is this may be the reason that caused the data loss?

I have tried a lot, but didn't find any clue to solve this problem.

Any help will be greatly appreciated.

Thanks,
Zheng

diabolo38, 2018-10-19:
You have to queue multiple xfer in advance doing it in callback is to late. Likely why you miss data as soon as system or bus is a bit buzy.

You may use multiple context or just one to handle multiple device I guess multiple may be slight better in multithtead app by minizing interlocking on single context.

zheng-rong, 2018-10-22:
@diabolo38 Thanks for your reply. 
I queued multiple xfer and then submit them before infinite loop. In the callback function, I just process data and then resubmit the xfer. I think this is "queue multiple xfer in advance" as you said, right?

DO you mean the multi-thread's  lock-unlock operation on one context will lower the xfer performance? But if I launch 2 app instances with independent camera, I got the same issue. How to explain this? I thought in each app instance, the process has its own context. Is my understanding right?

Best regards,
Zheng

diabolo38, 2018-10-22:
Then you do right. Has you get support on the mail list I want add  more here.

zheng-rong, 2018-10-23:
Hi @diabolo38 
Yes, This is the mailing list address: https://sourceforge.net/p/libusb/mailman/message/36445094/

Thanks,
Zheng

closed 2019-01-17 by mcuee