libusb/libusb #488: What's "libusb: error [submit_bulk_transfer] submiturb error -1 errno=2" ?

gachakism, 2018-10-30:
Hello, I'm very beginner of ROS and Linux user.
Now I have to make the existing ROS program work.

But when I try to debug that program, I always got streaming "libusb: error [submit_bulk_transfer] submiturb error -1 errno=2" errors.

I thought it caused from ROS firstly.
But that error message implies from libusb.

I already have referred https://stackoverflow.com/questions/44122912/libusb-error-submit-bulk-transfer-submiturb-failed-error-1-errno-2-what-am
however it doesn't work.

I attached my codes.
[emg_msg.zip](https://github.com/libusb/libusb/files/2527549/emg_msg.zip)
[emg_usb.zip](https://github.com/libusb/libusb/files/2527550/emg_usb.zip)

Do you know what's this error? or where it comes from?

My configuration is below.
・Ubuntu 16.04
・ROS ver. kinertic
・LibUsb-1.0.0

gachakism, 2018-11-06:
A little bit progress.
I run below test code. I got very weird error.

````
#include <stdio.h>
    #include <stdlib.h>
    #include <libusb-1.0/libusb.h>
 #pragma comment(lib,"libusb-1.0.lib")

 void die(char* s)
 {
    printf("Error: ");
    puts(s);
    exit(-1);

libusb_exit(NULL);
 }

 int main()
 {
    int ret;

    //initialize
    if ( libusb_init(NULL) < 0 )
        die("libusb_init");
    printf("Init\n");

    //Setting of debug level
    libusb_set_debug(NULL, LIBUSB_LOG_LEVEL_WARNING);
    printf("Set debug\n");

    //Open the device with VID & PID
    libusb_device_handle *devh = libusb_open_device_with_vid_pid(NULL, 0x04B4, 0xF131);
    if ( devh == NULL)
        die("libusb_open_device_with_vid_pid");
    printf("libusb_open_device_with_vid_pid\n");

    //Claim the interface of the device
    //Without doing, you will get the error like below
    //libusb: error [winusbx_submit_bulk_transfer] unable to match endpoint to an open interface - cancelling transfer

    ret = libusb_claim_interface(devh,0);
    if ( ret == 0 )
    {
        printf("OK\n");
    }else if ( ret == LIBUSB_ERROR_NOT_FOUND )
        die("libusb_claim_interface : LIBUSB_ERROR_NOT_FOUND");
    else if ( ret == LIBUSB_ERROR_BUSY )
        die("libusb_claim_interface : LIBUSB_ERROR_BUSY");
    else if ( ret == LIBUSB_ERROR_OVERFLOW )
        die("llibusb_claim_interface : LIBUSB_ERROR_NO_DEVICE");
    else
        die("libusb_claim_interface : UNKNOWN %d");
    printf("libusb_claim_interface\n");

    //---

    unsigned char data[64]={0x81};
    int data_len=1;

    //---

    //Submit bulk to EP=1. Time out  for 1000ms
    ret = libusb_bulk_transfer(devh, LIBUSB_ENDPOINT_OUT | 1, data, sizeof(data), &data_len, 1000);
    if ( ret == 0 )
    {
        printf("Received : %d Bytes\n", data_len);
        for ( int i = 0; i<data_len; i++ )
        {
            printf("%02X ", data[i]);
        }
    } else if ( ret == LIBUSB_ERROR_TIMEOUT )
        die("libusb_bulk_transfer : LIBUSB_ERROR_TIMEOUT");
    else if ( ret == LIBUSB_ERROR_PIPE )
        die("libusb_bulk_transfer : LIBUSB_ERROR_PIPE");
    else if ( ret == LIBUSB_ERROR_OVERFLOW )
        die("libusb_bulk_transfer : LIBUSB_ERROR_OVERFLOW");
    else if ( ret == LIBUSB_ERROR_NO_DEVICE )
        die("libusb_bulk_transfer : LIBUSB_ERROR_NO_DEVICE");
    else
        die("libusb_bulk_transfer : UNKNOWN");

    printf("\nSent\n");

    //---

    //Receiving bulk form EP=1. Time out for 1000ms
    ret = libusb_bulk_transfer(devh, LIBUSB_ENDPOINT_IN|1, data, sizeof(data), &data_len, 1000);
    if(ret == 0)
    {
        printf("Received : %d Bytes\n",data_len);
        for(int i=0;i<data_len;i++ )
        {
            printf("%02X ",data[i]);
        }
    }else if ( ret == LIBUSB_ERROR_TIMEOUT )
        die("libusb_bulk_transfer : LIBUSB_ERROR_TIMEOUT");
    else if ( ret == LIBUSB_ERROR_PIPE )
        die("libusb_bulk_transfer : LIBUSB_ERROR_PIPE");
    else if ( ret == LIBUSB_ERROR_OVERFLOW )
        die("libusb_bulk_transfer : LIBUSB_ERROR_OVERFLOW");
    else if ( ret == LIBUSB_ERROR_NO_DEVICE )
        die("libusb_bulk_transfer : LIBUSB_ERROR_NO_DEVICE");
    else
        die("libusb_bulk_transfer : UNKNOWN");

    printf("\nReceived\n");

    //Close libusb
    libusb_close(devh);
    libusb_exit(NULL);

    printf("Done\n");
    return 0;}

``

After I run,

> Init
> Set debug
> libusb_open_device_with_vid_pid
> OK
> libusb_claim_interface
> libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=2
> Error: libusb_bulk_transfer : UNKNOWN
 
I got OK and  libusb_bulk_transfer : UNKNOWN at the same time.
I don't know why.

I can understand error in my question is concerning about LIBUSB_IO_ERROR and if I make my libusb down-grade I may solve this problem.
How do you think about solving this problem? and Which version of libusb will I run this code?

My configuration is here
My configuration is below.
・Ubuntu 16.04
・libusb-1.0.20

mcuee, 2019-01-17:
I think this has nothing to do with libusb version. You need to understand the communication protocol of the device in order to use libusb. Please use the mailing list for further support,

closed 2019-01-17 by mcuee