libusb/libusb #489: Libusb hotplug callback fired before kernel recognize my device

waeldraoui, 2018-11-03:
Hello,

I'm using libusb on an embedded linux system to detect plug/unplug of my BLE dongle.
I registered my libusb callback in the initialization of my thread then, for architecture issues, i'm handling libusb events in another thread. 

```
// in init function
  res = libusb_hotplug_register_callback(NULL, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED |
                                        LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, 0, 0xFFFF, 0x0001,
                                        LIBUSB_HOTPLUG_MATCH_ANY, hotplug_callback, dongle_info,
                                        &handle);
    if (res != LIBUSB_SUCCESS) {
    libusb_exit(NULL);
    return -1;
  }
  
  static pthread_t tid_usb = 0;
  pthread_attr_t attr;
  pthread_attr_init(&attr);
  pthread_create( &tid_usb, &attr, event_handler, dongle_info );
// continue init of my structs
```

Once I plug my dongle, the callback is fired with the appropriate flag. The problem is that the callback is fired before my kernel recognize the dongle. So when I call the function `libusb_claim_interface (libusb_device_handle *dev, int interface_number)` I recieve the error -5 (i.e LIBUSB_ERROR_NOT_FOUND).

I used to implement a retry mechanism that close the libusb device and reopen it but that dosen't solve the problem.
The only workaround that works is putting a usleep() before initializing the device but it's very ugly and may cause other problems. So the question is, do libusb provide a solution for this problem ? 
If no, has someone a better solution than mine ?

Regards,
Wael

LudovicRousseau, 2018-11-03:
I had a similar, but inverse, problem. On macOS the device was detected by the OS hotplug mechanism but was not yet ready for libusb.
I used a retry mechanism and it works fine for me.
I do not use the libusb hotplug mechanism.

https://salsa.debian.org/rousseau/CCID/blob/master/src/ccid_usb.c#L779

hjelmn, 2018-11-04:
Odd on both fronts. If using udev the callback doesn’t fire until the device is created. On macOS the semantic should be similar. Never seen a case where a device matched the filter but is not immediately available (both using libusb and my own hotplug code). Does this happen with a specific macOS version or type of device? Is retrying the only workaround that seems to work?

It might be worth adding a retry loop to libusb when the device was detected by hotplug.

-Nathan

LudovicRousseau, 2018-11-04:
I don't know if it is related to a specific version of macOS. I could try to remove the retry loop to see if the problem is still present in High Sierra or Mojave.

I have the problem with a USB smart card driver. The driver is loaded by the macOS component ``/System/Library/CryptoTokenKit/com.apple.ifdreader.slotd/Contents/MacOS/com.apple.ifdreader`` when the USB device is connected. I don't know exactly what OS component is detecting the device.

waeldraoui, 2018-11-05:
> Odd on both fronts. If using udev the callback doesn’t fire until the device is created. On macOS the semantic should be similar. Never seen a case where a device matched the filter but is not [...]

@hjelmn Is there an example of a hotplug callback using udev so I can inspire a solution from it ?

dickens, 2020-01-13:
On Linux, the uevent that udev is listening for and the creation of the device node (e.g. /dev/bus/usb/XXX/YYY) can race with each other.  Generally when this occurs you will get an error when calling `libusb_open()` because the node either doesn't exist yet or doesn't have the appropriate permissions.  Your case appears to be slightly different since you were able to successfully open the device.

For Linux, `libusb_claim_interface()` returns `LIBUSB_ERROR_NOT_FOUND` only when the kernel returns `ENOENT` as the error code.  This only occurs when the desired interface cannot be found in the device's current configuration.  If the device is unconfigured, no interfaces will be able to be claimed.  Can you provide the output of dmesg when this issue is observed?  You may also want to try calling `libusb_get_configuration()` prior to claiming the interface to see what you get.

dickens, 2020-08-18:
Closing this since there has been no response for over 6 months.  Feel free to provide additional information.

closed 2020-08-18 by dickens

ProfessorLongBeard, 2024-08-14:
> Hello,
> 
> I'm using libusb on an embedded linux system to detect plug/unplug of my BLE dongle. I registered my libusb callback in the initialization of my thread then, for architecture issues, i'm handling libusb events in another thread.
> 
> ```
> // in init function
>   res = libusb_hotplug_register_callback(NULL, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED |
>                                         LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, 0, 0xFFFF, 0x0001,
>                                         LIBUSB_HOTPLUG_MATCH_ANY, hotplug_callback, dongle_info,
>                                         &handle);
>     if (res != LIBUSB_SUCCESS) {
>     libusb_exit(NULL);
>     return -1;
>   }
>   
>   static pthread_t tid_usb = 0;
>   pthread_attr_t attr;
>   pthread_attr_init(&attr);
>   pthread_create( &tid_usb, &attr, event_handler, dongle_info );
> // continue init of my structs
> ```
> 
> Once I plug my dongle, the callback is fired with the appropriate flag. The problem is that the callback is fired before my kernel recognize the dongle. So when I call the function `libusb_claim_interface (libusb_device_handle *dev, int interface_number)` I recieve the error -5 (i.e LIBUSB_ERROR_NOT_FOUND). Note that libusb_open and libusb_kernel_driver_active functions don't return errors.
> 
> I used to implement a retry mechanism that close the libusb device and reopen it but that dosen't solve the problem. The only workaround that works is putting a usleep() before initializing the device but it's very ugly and may cause other problems. So the question is, do libusb provide a solution for this problem ? If no, has someone a better solution than mine ?
> 
> Regards, Wael

How are any of you even getting the callback working? I setup a thread for the callback polling but the callback never gets called