libusb/libusb #490: WinUSB backend stutters on bulk IN requests

bsccara, 2018-11-05:
I'm working on a Java app that uses LibUsb 1.0.22 through Usb4Java 1.3.0. I need to get data through bulk transfers from a device that implements the WinUSB spec. Since the data is streamed from the device at the maximum possible speed I keep 4 IN transfers in-flight at all times. The PC (same for both OS) has a A4-5000 CPU, so its no powerhouse, and the device uses Full Speed, so no speed demon either.
When running on Kubuntu 18.04 I can sustain the maximum throughoutput, 12~13 64-byte packets per frame, continuously. But on Windows 8.1 I'm getting periods of around 1 frame with no USB traffic, apart from SOFs, after the end of most transfers (each 128K). This was checked on the wire, not on the PC.
Reading the WinUSB spec I noticed that SetPipePolicy can accept a RAW_IO policy. From my understanding if that policy is not set WinUSB will queue all IN requests and feed them one at a time to the lower layers, therefore negating the purpose of having multiple requests in-flight. The docs claim that setting that policy makes it pass the requests along immediately, prevent that issue.
Since LibUSB's WinUSB backend does not set that policy does that mean that the code cannot support RAW_IO or can I enable it some other way on my app (through JNA probably) ?

mcuee, 2019-01-17:
You may want to ask in libusb-devel mailing list to get better help.

mcuee, 2021-01-16:
https://marc.info/?t=145424689900001&r=1&w=2

>From Tim
I think a good argument could be made that libusb should set RAW_IO unconditionally.  Libusb itself does much of the buffer management that gets triggered in WinUSB when RAW_IO is turned off

mcuee, 2021-01-16:
Still it will make libusb Windows which is already complex even more complex.
>From Chris
Using RAW_IO introduces certain complexities for libusb. In
particular, if the provided transfer buffer is not a multiple of the
endpoint max packet size, we have two options:

1) Round the buffer size down to a max packet size multiple. If the
buffer is smaller than the endpoint max packet size, libusb has to
allocate a buffer.
2) Round the buffer size up to the next max packet size multiple and
allocate a buffer of that size.

Now if we have to allocate and the transfer completes with more data
than the user's buffer will allow, we have to keep the excess data
until the next transfer.

Then there is the case where the user's buffer is greater than
MAXIMUM_TRANSFER_SIZE. Now the Windows backend is getting into the
business of breaking up the transfer into multiple smaller ones.

mcuee, 2021-01-26:
Close this for now as we can not change it for now. Feel free to re-open if there are new proposals.

PetteriAimonen, 2022-02-11:
What about this: add option to `libusb_set_option()` to enable `RAW_IO`. When it is enabled, throw error if transfer buffer is not a multiple of packet size.

The current lack of `RAW_IO` makes libusb on Windows unusable with FX2. The small buffer size on device side causes it to drop data during the transfer swap. E.g. sigrok is using their own patched libusb version for this reason.

PetteriAimonen, 2022-02-11:
Actually, even simpler is to enable `RAW_IO` dynamically when the transfer size is known to be multiple of 512 (which is divisible by all packet sizes).

I have implemented that here: https://github.com/devEmbedded/libusb/tree/dev_raw_io_2
Prebuilt binary for testing: https://github.com/devEmbedded/libusb/actions/runs/1828706082

diabolo38, 2022-02-11:
> Actually, even simpler is to enable `RAW_IO` dynamically when the transfer size is known to be multiple of 512 (which is divisible by all packet sizes).
wrong assumption if device is super speed it may be 1024 !

mcuee, 2022-02-12:
> What about this: add option to `libusb_set_option()` to enable `RAW_IO`. When it is enabled, throw error if transfer buffer is not a multiple of packet size.

I tend to think this is better.

mcuee, 2022-02-12:
> The current lack of `RAW_IO` makes libusb on Windows unusable with FX2. The small buffer size on device side causes it to drop data during the transfer swap. E.g. sigrok is using their own patched libusb version for this reason.

Can you point out where is the libusb fork from Sigrok project?

PetteriAimonen, 2022-02-12:
@mcuee Sigrok's gitweb seems to be offline currently, but the patch they use is here (unconditionally enables RAW_IO): https://github.com/sigrokproject/sigrok-util/blob/master/cross-compile/mingw/mxe_fixes.patch#L225

tormodvolden, 2022-02-14:
So does sigrok use this on many different devices or is it targeted to one or a few of them?

diabolo38, 2022-02-14:
> So does sigrok use this on many different devices or is it targeted to one or a few of them?

I guess it is few (a lot) . Most of these low cost usb "scope/lsa" are often just an  FX2 parallel 8/16 bit  gpio to bulk ep with limited fifo size why it goes bad w/o assync xfer as soon you raise sampling freq. They should have use iso not bulk.

PetteriAimonen, 2022-02-15:
@tormodvolden Without RAW_IO, the WinUSB backend only has one request active at a time. This in turn causes a pause in transmission when the buffer is switched. So it affects any device that has less than about 5 ms worth of buffer space.

@diabolo38 AFAIK isochronous on WinUSB backend doesn't work very well either, but I'm not sure if that is related to this or not.

diabolo38, 2022-02-15:
@[tormodvolden](https://github.com/tormodvolden) hum to me 5ms is over kill and i never get so bad back to back pause time in between request even with the synch api , it's more in range 300us to 1ms?  i almost never use winusb backend only LibUsbK that may explain part of my observation.

For iso/bulk  it's to me more on the device f/W  and initial  technical choice to use bulk vs iso but whenever they are strong latency and bandwidth requirement and constrain shall be clear => ISO, bulk may do but it's risky but often so much simpler to handle (why the bad choice?)

It's  true that so far iso support under window is far behind (when none) to other OS.

PetteriAimonen, 2022-02-15:
@diabolo38 It seems the buffer switch latency varies a lot between PCs. Maybe it is related to other drivers running in Windows kernel? I agree that less than 500 us pauses is the most common case and the over 1 ms pauses happen very rarely, making it a pain to debug.

diabolo38, 2022-02-15:
PC but not only,  what device on the same usb  controller (look at device manager, view by connection)  can make huge difference 
it's sometime enough to have irq or isoch device (mice/keyboard/sound/webcam.) to affect a lot bulk perf.

By the way what are the size for each request ? if too small it's just make problem worst and want help in any sort if the data capture isn't use live.  Even it's use live  fitting size to get 50 req/second want be much visible to  user in UI , limit is at 2MB/4MB can't remember exactly but you won't need that much on usb 2 .

tormodvolden, 2022-02-15:
OK, from what I really meant with my question, FX2 devices are "one" kind of device :) So their patch might create issues on other devices that they don't care about, I suppose. To me it sounds like WinUSB is not a great backend for this purpose. What does it look like for other backends (usbdk) or drivers (libusbK)?

mcuee, 2022-04-01:
Related discussion for isoc transfer: https://github.com/libusb/libusb/pull/1069

martinling, 2022-04-09:
I remember running into this when I was working on sigrok, and I expect it's also the reason that HackRF can't stream at full sample rate on Windows.

For both projects, I think the `libusb_set_option` approach would be fine if an automatic solution isn't practical.

PetteriAimonen, 2022-04-09:
I really wish this would get solved for WinUSB. There are already two good solutions, one with some theoretical risk of regressions, though none has been seen in practice - and another solution that has zero risk of regression.

If there is anything I can do to help get this merged, just let me know!

mcuee, 2022-04-10:
> If there is anything I can do to help get this merged, just let me know!

@PetteriAimonen I think it just need more reviews and testing report.

mcuee, 2022-04-10:
> I remember running into this when I was working on sigrok, and I expect it's also the reason that HackRF can't stream at full sample rate on Windows.
> 
> For both projects, I think the `libusb_set_option` approach would be fine if an automatic solution isn't practical.

Exactly. But there are some complexities as mentioned by Chris in #149.
* #149

martinling, 2022-04-10:
> Exactly. But there are some complexities as mentioned by Chris in #149 if we want to have a generic libusb_set_option approach to enable RAW_IO policy for bulk transfer and isoc transfer.

The discussion in #149 is about turning on RAW_IO unconditionally. None of the problems Chris is talking about there would apply to a `libusb_set_option` approach.

A program would have to explicitly opt-in to the use of RAW_IO by setting the option. When doing so, it would be promising to ensure that all transfer buffers it submits will be a multiple of their endpoint's maximum packet size.

PetteriAimonen, 2022-04-10:
We can also do combination of both: when option is enabled, enable RAW_IO when transfer size is a multiple. That could enable testing applications for compatibility and then later allow enabling the option by default for all applications. For testing an environment variable could be a nice addition to `libusb_set_option`.

As for testing reports, it will be very difficult to get people for who libusb currently works to specifically install a modified libusb to verify it still works - there is no motivation.

Here is my suggestion that I can help implement:

- Add the `libusb_set_option(ctx, LIBUSB_OPTION_WINUSB_RAW_IO, 1)` that defaults to off. Add environment variable `LIBUSB_OPTION_WINUSB_RAW_IO` that enables this option, to make it possible to easily test with applications without modifying their code.
- When the option is enabled, enable `RAW_IO` for transfers that have transfer size a multiple of 1024, which covers everything from USB low-speed to USB super-speed.
- In the future, we can consider enabling the option by default.

No risk of regression, and also provides an update path in the future to deprecate the specialized per-platform option.

martinling, 2022-04-10:
I don't think RAW_IO can be safely applied to selected transfers - I've gone into detail on this [here](https://github.com/libusb/libusb/pull/1069#pullrequestreview-937231429).

PetteriAimonen, 2022-04-10:
@martinling I agree with your reasoning about varying transfer sizes. Not sure how common that is in practice, but it is a true issue that may not be possible to circumvent with fully automatic enabling.

martinling, 2022-04-10:
The shortcoming I see with the `libusb_set_option` approach is that it forces the setting to be applied at the context level rather than on an individual endpoint.

I can imagine scenarios in which it might be difficult to ensure all code that runs within a given context uses appropriately-sized buffers, when what is needed for performance is just to enable the use of RAW_IO for a single endpoint.

A better change would be to add a new `libusb_set_endpoint_option` mechanism to the API, which could allow for any future endpoint-specific options as well.

For now we would just need one option, the effect of which will be to tell libusb that all submitted transfers will have a buffer size that is a multiple of the endpoint's max packet size.

PetteriAimonen, 2022-04-10:
Annoying thing about adding new API is that it will take a long time to trickle down to the various libusb bindings. And depending on how they are implemented, they cannot even add it without risking incompatibility with older versions of libusb on different platforms.

One option could be to use [libusb_transfer_flags](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#ga1fb47dd0f7c209b60a3609ff0c03d56d) for this.

mcuee, 2022-04-10:
> A better change would be to add a new libusb_set_endpoint_option mechanism to the API, which could allow for any future endpoint-specific options as well.

That is not necessary. I think we can use libusb_set_option(ctx, WINUSB_RAW_IO, ep) or even libusb_set_option(ctx, WINUSB_RAW_IO, intf, altf, ep).

martinling, 2022-04-10:
> That is not necessary. I think we can use `libusb_set_option(ctx, LIBUSB_OPTION_WINUSB_RAWIO, ep)` or even `libusb_set_option(ctx, LIBUSB_OPTION_WINUSB_RAWIO, intf, altf, ep)`.

Oh, I'd missed that `libusb_set_option` is variadic. In that case yes, only the new flag is needed.

I think the arguments should be be a `libusb_device *` and the endpoint address. The device needs to be open with the relevant interface already set, for it to be possible to set RAW_IO. So the call would be:
```c
libusb_set_option(ctx, LIBUSB_OPTION_WINUSB_RAWIO, device, endpoint_address)
```

> This shoud not be difficult for bulk transfer. For isoc transfer, we need to consider the special properties of the High Speed and Super Speed USB isoc endpoints.

This doesn't involve any of the isochronous buffer rules. The RAW_IO flag is only applicable to bulk and interrupt endpoints, as specified [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-functions-for-pipe-policy-modification).

mcuee, 2022-04-10:
> This doesn't involve any of the isochronous buffer rules. The RAW_IO flag is only applicable to bulk and interrupt endpoints, as specified [...]

Oops, I mixed this up with something else. Last time there were three WinUSB pipe policy related to isoc transfer, but now it seems no mentioned any more.
http://libusbk.sourceforge.net/UsbK3/usbk_pipe_management.html
1) ISO_START_LATENCY
2) ISO_ALWAYS_START_ASAP
3) ISO_NUM_FIXED_PACKETS

Now I can only see the WinUSB isoc functions here.
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-set-up-to-use-windows-devices-usb

mcuee, 2022-04-10:
> I think the arguments should be be a `libusb_device *` and the endpoint address. The device needs to be open with the relevant interface already set, for it to be possible to set RAW_IO. So the call would be:
> 
> ```c
> libusb_set_option(ctx, LIBUSB_OPTION_WINUSB_RAWIO, device, endpoint_address)
> ```

Looks like a good one. Thanks.

mcuee, 2022-04-10:
BTW, so far for open source projects dealing with isoc transfer under libusb under Windows, I think libusbk is the best. It supports isoc transfer (FS, HS and SS) with both libusbk and WinUSB driver.
Ref: https://github.com/mcuee/libusbk

Performance testing results for SS:
https://github.com/mcuee/libusbk/tree/master/BmFW/CYPRESS_FX3

martinling, 2022-04-10:
> BTW, so far for open source projects dealing with bulk/isoc transfer under libusb under Windows, I think libusbk is the best. It supports isoc transfer (FS, HS and SS) with both libusbk and WinUSB driver.

As I understood it, libusbk uses its own kernel driver, and switching to that involves making users mess around changing drivers. If the libusbk backend can also use the default WinUSB kernel driver and gets better performance, then shouldn't whatever it's doing to achieve that should just be merged into libusb's WinUSB backend?

mcuee, 2022-04-10:
> As I understood it, libusbk uses its own kernel driver, and switching to that involves making users mess around changing drivers. If the libusbk backend can also use the default WinUSB kernel driver and gets better performance, then shouldn't whatever it's doing to achieve that just be merged into libusb's WinUSB backend?

The current libusb Windows backend already integrated libusbk.dll support long ago, with the exception of isoc transfer support for WinUSB.

1) If libusbk.dll is found, libusb Windows will use libusbk.dll to support devices with WinUSB, libusbk.sys and libusb0.sys driver.
2) If libusbk.dll is not found, libusb WIndows will not be able to support devices with libusbk.sys and libusb0.sys, only deviced with WinUSB driver will work.
3) Exception is that previously libusbk,dll did not support isoc transfer of WinUSB devices, therefore there are different code path for isoc transfer -- one for devices with libusbk.sys (libusbk.dll required) and one for WinUSB (with or without libusbk.dll)

The current libusb Windows backend is very complicated partially because of different code path.
1) WinUSB/libusbk/libusb0 support mentioned above
2) HID driver support
3) usbdk backend support (if usbdk is present and libusb_set_option specifies usbdk backend.

closed 2023-01-22 by tormodvolden