libusb/libusb #491: nullptr deref after get_ancestor() call in windows_winusb.c

Youw, 2018-11-06:
I'm getting crash, when trying to enumerate devices with libusb_get_device_list().
I was lucky to get a crash dump from a field, and here is where it led me:
In [this line of code](https://github.com/libusb/libusb/blob/e5a7bb548a5c3b04a086b62de043f02fdae338b6/libusb/os/windows_winusb.c#L833) libusb dereference a pointer, received from `get_ancestor()` function call:
```cpp
for (depth = 1; bus_number == 0; depth++) {
    tmp_dev = get_ancestor(ctx, devinst, &devinst);
    if (tmp_dev->bus_number != 0) {
    //  ^^^^^^^ - here is my crash
        bus_number = tmp_dev->bus_number;
        depth += _device_priv(tmp_dev)->depth;
    }
    libusb_unref_device(tmp_dev);
}
```

I'm not well familiar with this code, and not sure whether `tmp_dev` should be simply checked before dereferencing, or the logic should be changed more significant (and, maybe, higher in the call stack).

Here is a screenshot of some locals at the moment of crash:
[image]

I'm whiling to assist in any way I can, to find a core reason of this crash.
---
NOTES:
1) I don't have access to the machine where this happened, and don't know hardware configuration nor what USB devices connected (except one);
2) I cannot reproduce it - never happen for me or any of my team mates;
3) I have crash dump and debug symbols, but I cannot share them publicly.

mcuee, 2019-01-17:
You may want to ask in libusb-devel mailing list to get better help. I think you have to get to know the USB device connected in order to get better help.

Youw, 2019-01-17:
That crash was the only instance we've ever seen on some of our clients machines. (Crashdump collected automatically).
No more similar crashes since then.

I think at this stage, this ticket has more historical value.

Corjan85, 2019-04-12:
Hi,

I get exactly the same crash.

As a workaround I have changed the line to:
if ((tmp_dev != NULL) && (tmp_dev->bus_number != 0)) {

This seems to work, although I have no idea what kind of complications this might give.

It crashes on the Silex SX-DS-4000U2:
[Silex SX-DS-4000U2](https://www.silextechnology.com/sx-ds-4000u2)

I have the option to debug this device remotely with a customer. So please let me know if you want me debug any piece of code.

Corjan

frgmarket, 2020-08-03:
Hello,

this problem still reproduces. I get the same crash when using Silex DS-600 device, using libusb 1.0.23 & Windows 10.

mcuee, 2020-08-04:
@frgmarket Please help to post the debug log. Thanks.

mcuee, 2020-08-04:
@Corjan85 Please post the debug log if you are still able to do that. Thanks.

mcuee, 2020-08-04:
https://www.silextechnology.com/sx-ds-4000u2

The device seems to be a very special device, kind of USB server/proxy on the gigabit Ethernet.

frgmarket, 2020-08-05:
@mcuee here is debug log from running listdevs up until crash happens. 1 USB device (AX88772) is connected to Silex USB server. Hope this helps.

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000001] [00000cfc] libusb: debug [libusb_init] created default context
[ 0.000109] [00000cfc] libusb: debug [libusb_init] libusb v1.0.23.11523
[ 0.000192] [00000cfc] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000204] [00000cfc] libusb: debug [windows_init_clock] hires timer frequency: 10000000 Hz
[ 0.000217] [00000cfc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006455] [00000cfc] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.007112] [00000cfc] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.008062] [00000cfc] libusb: info [windows_init] UsbDk backend is not available
[ 0.008097] [00000cfc] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.008114] [00000cfc] libusb: debug [libusb_get_device_list]
[ 0.015516] [00000cfc] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'ROOT'
[ 0.015539] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [1D]
[ 0.015827] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [4C]
[ 0.015906] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [4D]
[ 0.015984] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [4E]
[ 0.016070] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [4F]
[ 0.016155] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [50]
[ 0.016233] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.016310] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [52]
[ 0.016392] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.016471] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.016548] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [55]
[ 0.017012] [00000cfc] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017075] [00000cfc] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017087] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [3E]
[ 0.017259] [00000cfc] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.017312] [00000cfc] libusb: debug [get_api_type] matched driver name against HID API
[ 0.017323] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [3F]
[ 0.017445] [00000cfc] libusb: debug [get_api_type] driver(s): AX88772
[ 0.017498] [00000cfc] libusb: debug [winusb_get_device_list] allocating new device for session [40]
[ 0.017744] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_3A39&SUBSYS_50041458&REV_00\3&13C0B0C5&0&D2' bus number 1
[ 0.017828] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_3A3A&SUBSYS_50061458&REV_00\3&13C0B0C5&0&EF' bus number 2
[ 0.017878] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_3A3C&SUBSYS_50061458&REV_00\3&13C0B0C5&0&D7' bus number 3
[ 0.017928] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_3A35&SUBSYS_50041458&REV_00\3&13C0B0C5&0&E9' bus number 4
[ 0.017977] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_3A36&SUBSYS_50041458&REV_00\3&13C0B0C5&0&EA' bus number 5
[ 0.018026] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_1033&DEV_0194&SUBSYS_50071458&REV_03\4&115A8633&0&0010' bus number 6
[ 0.018083] [00000cfc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'ROOT\SYSTEM\0001' bus number 7
[ 0.018096] [00000cfc] libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ROOT\SYSTEM\0001'
```

mcuee, 2020-08-05:
Looks like a Silex problem. Please point to them this ticket and ask them to fix their FW or driver.

mcuee, 2020-08-05:
On the other hand, if someone can submit a patch to work around this issue in libusb, it is okay to submit pull request for review.

dickens, 2020-08-14:
This driver is broken in a major way.  I installed it locally to see how it presents itself to the system and there are a number of issues.  I'll work on making libusb more robust to such buggy virtual devices, but there's no way it will work with libusb without the vendor fixing the driver.

commit aa6ec9d9 (2020-08-17): Windows: Prevent NULL pointer dereference when ancestor is missing

A buggy virtual USB device driver can cause the device enumeration
process to fail during the init_device() function when trying to
determine the bus number of the device. Guard against this by checking
that the ancestor device was actually found and skipping the bogus
device when there is no ancestor.

Closes #491
closed 2020-08-17 by dickens