libusb/libusb #492: Could not list devices at MacOS 10.13.6 High Sierra

timdesi, 2018-11-07:
Could not list devices at MacOS 10.13.6 High Sierra. 
Try latest code with fix for https://github.com/libusb/libusb/pull/428, but nothing changes.

Sample console output

UsbHub Bus 26 Device 0 : VendorId(0x05ac) ProductId(0x8006) null null
UsbHub Bus 26 Device 1 : VendorId(0x8087) ProductId(0x0024) null null

On MacOS 10.12.6 everything works us expected.

Sample console output

UsbHub Bus 20 Device 0 : VendorId(0x05ac) ProductId(0x8007) null null
 -- UsbDevice Bus 20 Device 5 : VendorId(0x046d) ProductId(0xc31c) Logitech USB Keyboard
 -- UsbDevice Bus 20 Device 4 : VendorId(0x17ef) ProductId(0x600e) PixArt Lenovo USB Optical Mouse
 UsbHub Bus 20 Device 3 : VendorId(0x0a5c) ProductId(0x4500) Apple Inc. BRCM20702 Hub
 -- UsbDevice Bus 20 Device 7 : VendorId(0x05ac) ProductId(0x8289) Apple Inc. Bluetooth USB Host Controller
 -- UsbDevice Bus 20 Device 2 : VendorId(0x05ac) ProductId(0x8242) Apple, Inc. IR Receiver
 -- UsbDevice Bus 20 Device 10 : VendorId(0x147b) ProductId(0x5120) ABB STOTZ-KONTAKT GmbH KNX-USB Interface (MDRC)
 -- UsbDevice Bus 20 Device 9 : VendorId(0x10c4) ProductId(0x8293) Silicon Labs Telegesis USB Device

mcuee, 2020-07-29:
What is the real use case here? Is this anomaly still valid now?

mcuee, 2020-12-07:
FYI, for the latest version of libusb under my Mac Mini M1, the output is like this. The root hub is no longer listed.

```
mcuee@mcuees-Mac-mini libusb % sw_vers
ProductName:	macOS
ProductVersion:	11.0.1
BuildVersion:	20B29

mcuee@mcuees-Mac-mini libusb % ./examples/listdevs 
1915:1025 (bus 2, device 13) path: 2.5.6
04b4:8613 (bus 2, device 12) path: 2.5.2
04d8:0033 (bus 2, device 11) path: 2.5.1
0ac8:3420 (bus 2, device 10) path: 1.4
0bda:5411 (bus 2, device 9) path: 1.1
0a5c:21e8 (bus 2, device 8) path: 2.7
046d:c52b (bus 2, device 7) path: 2.2
1a40:0201 (bus 2, device 6) path: 2.5
05e3:0749 (bus 2, device 5) path: 5.1.1
0bda:5411 (bus 2, device 4) path: 1
0bda:0411 (bus 2, device 3) path: 5.1
1a40:0201 (bus 2, device 2) path: 2
0bda:0411 (bus 2, device 1) path: 5
```

mcuee, 2020-12-07:
@timdesi I will close this issue for now. Please re-open if you have new info.

closed 2020-12-07 by mcuee