libusb/libusb #494: add cmake support

seker, 2018-11-09:
Hello, I wish this project can add CMakeLists.txt file. 
So that, we can dev in almost every IDE. (I am using CLion)
Thanks~

LudovicRousseau, 2018-11-09:
I think this issue is a duplicate of https://github.com/libusb/libusb/issues/322

hjelmn, 2018-11-09:
Yup. And unless something has changed this will be closed as well.

LudovicRousseau, 2018-11-13:
Thanks for the PR in https://github.com/libusb/libusb/pull/495

The problem is that no libusb core developer has the knowledge/time/motivation/etc. to maintain a cmake configuration on the long term.

seanm, 2019-01-02:
+1 for the general idea of supporting CMake.

There is a lot of tooling around CMake that is quite useful.  ex: CMake can output compile commands in json:

https://cmake.org/cmake/help/latest/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html

and that format can be read by cppcheck and clang-tidy, tools which could be used to check the libusb codebase for issues.

hjelmn, 2019-01-02:
Not really that useful. Bear can do the same for any project. Autotools works well and is here to stay in the near term. I am open to changing if cmake actually offers something that isn't already available as the lack of bootstrapping is a big issue with cmake. I do not like the idea of requiring anyone who build from a tarball to need to install another dependency.

closed 2019-01-02 by hjelmn

TomzBench, 2019-12-05:
You can do something like 

include_external_msproject(mslib path/to/proj.vsxproj)
add_dependencies(mylib mslib)

See here:
https://cmake.org/cmake/help/latest/command/include_external_msproject.html

Youw, 2023-08-06:
For reference, help those who are trying to google this one: now there is a community-supported https://github.com/libusb/libusb-cmake