libusb/libusb #497: LGPL license link exception

nxpfrankli, 2018-11-27:
We are working on a BSD license tools, which use libusb.  we want to static link libusb in windows, so customer just need download a portable executive file. 

In libgit2 have linking exception.  Is it possible to add the similar exception in libusb projuct? 

https://github.com/libgit2/libgit2/blob/master/COPYING

----------------------------------------------------------------------

			LINKING EXCEPTION

 In addition to the permissions in the GNU General Public License,
 the authors give you unlimited permission to link the compiled
 version of this library into combinations with other programs,
 and to distribute those combinations without any restriction
 coming from the use of this file.  (The General Public License
 restrictions do apply in other respects; for example, they cover
 modification of the file, and distribution when not linked into
 a combined executable.)

----------------------------------------------------------------------

LudovicRousseau, 2018-11-27:
libusb uses the GNU **LESSER** GENERAL PUBLIC LICENSE, not the GNU GENERAL PUBLIC LICENSE as used by libgit2.

I think this is a non-issue.

nxpfrankli, 2018-11-27:
**LESSER** GPL still not allowed static link,  only allow dynamical link

LudovicRousseau, 2018-11-27:
I don't think so. Please point the license chapter that says that.

hjelmn, 2018-11-27:
Yeah, my understanding is that the LGPL allows even static linking. That exception in libgit2 is worded in the worst way. It allows modifications to the embedded library without ever contributing back.

nxpfrankli, 2018-11-27:
Our legal team said software need be GPL\LGPL if static link with LGPL library.  Can you help high light section which said allow static link into NO GPL's software?

hjelmn, 2018-11-27:
From wikipedia:

```
Essentially, if it is a "work that uses the library", then it must be possible for the software to be linked with a newer version of the LGPL-covered program. The most commonly used method for doing so is to use "a suitable shared library mechanism for linking". Alternatively, a statically linked library is allowed if either source code or linkable object files are provided.[2]
```

This means someone can request either the original libusb.a you used to link OR the source code used to generate it.

hjelmn, 2018-11-27:
At least that is the way I read it.

nxpfrankli, 2018-11-27:
Thanks, I will ask our legal\license team to comment this.  Anyway, it is good news.

nxpfrankli, 2018-11-29:
https://www.gnu.org/licenses/gpl-faq.html#LGPLStaticVsDynamic

closed 2018-11-29 by nxpfrankli