libusb/libusb #499: using udev for hotplugin, but it shows not find libudev.h

VitoHed, 2018-11-30:
but I use libudev, but it doesn't build success in NDK. It shows not find libudev.h, and I don't know how to build it. anybody can tell me ?

My android.mk is:

include $(CLEAR_VARS)
#BUILD_SHARED_LIBRARY/BUILD_EXECUTABLE/BUILD_STATIC_LIBRARY
LOCAL_SRC_FILES := \
  $(LOCAL_PATH)/usr/main.c \
  $(LOCAL_PATH)/libusb/os/linux_udev.c \
  $(LOCAL_PATH)/libusb/core.c \
  $(LOCAL_PATH)/libusb/descriptor.c \
  $(LOCAL_PATH)/libusb/hotplug.c \
  $(LOCAL_PATH)/libusb/io.c \
  $(LOCAL_PATH)/libusb/sync.c \
  $(LOCAL_PATH)/libusb/strerror.c \
  $(LOCAL_PATH)/libusb/os/linux_usbfs.c \
  $(LOCAL_PATH)/libusb/os/poll_posix.c \
  $(LOCAL_PATH)/libusb/os/threads_posix.c \
  $(LOCAL_PATH)/libusb/os/linux_netlink.c
  
LOCAL_MODULE_TAGS := optional

LOCAL_C_INCLUDES += \
		    $(LOCAL_PATH)/usr \
		    $(LOCAL_PATH)/libusb \
		    $(LOCAL_PATH)/libusb/os 
		    
LOCAL_CFLAGS := -DRIL_SHLIB -D_DEBUG -DANDROID_DEBUG -DPROC_VERSION=$(MYVERSION_STRING) -Wno-psabi

LOCAL_LDLIBS := -L$(LOCAL_PATH)/../systemlibs -llog -ludev

LOCAL_MODULE := hotplugtest

LOCAL_CFLAGS += -pie -fPIE 
LOCAL_LDFLAGS += -pie -fPIE 

include $(BUILD_EXECUTABLE)

hjelmn, 2019-01-09:
udev is not available on Android from what I understand. In that case you need to use --disable-udev. That will make it fall back on netlink.

mcuee, 2020-01-06:
I think we can close this.

BTW, libusb is now working under Android with Termux.
https://github.com/termux/termux-packages/blob/master/packages/libusb/build.sh

closed 2020-01-06 by mcuee