libusb/libusb #501: Wasm Support

MelbourneDeveloper, 2018-12-05:
WebUSB tanked.

Please get this library running in Wasm so we can get USB support in all browsers.

MelbourneDeveloper, 2018-12-05:
https://developers.google.com/web/updates/2018/03/emscripting-a-c-library

MelbourneDeveloper, 2018-12-05:
https://github.com/kripken/emscripten/issues/7610

LudovicRousseau, 2018-12-05:
libusb API is in C and documented at http://libusb.sourceforge.net/api-1.0/

If you want to use libusb from another language you have to write the wrapper or ask/pay someone else to do it for you.
Have a look at "Language Bindings" https://github.com/libusb/libusb/wiki#language-bindings for support of other languages provided by other projects.

MelbourneDeveloper, 2018-12-05:
Thanks. Yeah, it turns out that my understanding of wasm was incorrect. It's not really possible because wasm is sandboxed with the same functionality as JavaScript.

RReverser, 2022-01-20:
For what it's worth, I've just published an article on how I ported libusb to the web with WebAssembly and WebUSB: https://web.dev/porting-libusb-to-webusb/

If there's interest (and after more people try out my fork and report issues), I'm happy to upstream the backend to libusb too (I've intentionally created it in a way that integrates with libusb in the same way as any other backends).

mcuee, 2022-01-21:
Maybe you can just create a pull request here and then your codes will probably get more testers.

RReverser, 2022-01-21:
That sounds reasonable too. I'll submit something soon.

closed 2022-06-26 by tormodvolden

ConProgramming, 2024-12-22:
Hi @RReverser or @tormodvolden, is there any up to date guide on how to use libusb with webusb?

RReverser, 2024-12-22:
Yes, the blog post above.

commit ff8fe939 (2022-06-26): Add Emscripten backend for WebAssembly + WebUSB support

Fixes #501
Closes #1057