libusb/libusb #511: Seg Fault when libusb_exit() with libusb_init();

notlistening, 2018-12-16:
During some testing we found a senario where the library would not initalised correctly.

When running libusb in a docker container on armhf with QEMU 2.x as version of QEMU doesn't support netlink. 

So calling libusb_init() would fail and then it would call libusb_exit() to cleanup.

However this causes a segfault. This can be simplied by calling libusb_exit() first. 

I have not stepped through the code to see exactly where the issue is but i'd imagine is it a NULL pointer somewhere. 

Will post a patch if I get time to look at it. 

Also this behaviour/bug can be seen when other software fails in this way like pyusb.

mcuee, 2019-01-17:
Can you reproduce this issue without QEMU? I do not think this is a libusb issue.

hjelmn, 2019-04-05:
@mcuee We should probably strengthen the documentation of libusb_exit to indicate that it should not be called except after libusb_init is successfully invoked.

mcuee, 2019-04-06:
I agree that documentation can be improved.

commit a5624b22 (2020-01-13): core: Do not attempt to destroy a default context that doesn't exist

Calling libusb_exit(NULL) when a successful call to libusb_init(NULL)
has not been made results in a segmentation violation. This is
definitely a user error, but we can easily guard against it.

Closes #511
closed 2020-01-13 by dickens