libusb/libusb #512: Control Write Vendor Request Issue

alexanderplanar, 2018-12-17:
Greetings!
 I came accross a problem with the libusb driver using synchronous Control-OUT transfers (Vendor Write Requests) ( libusb_control_transfer with non-zero timeout).
 I try to implement a simple Request-Processing protocol in the device based on NXP LPC1758 MCU. This hardware has a special indicator bit (Packet Overwritten) which is used to intercept some situations when the unprocessed SETUP packet is being overwritten by the newest arrived packet. 
By the way, I had to peek the SETUP packet first without confirmation to preprocess it, and then to confirm it by clearing the hardware buffer. This point is the main potentially moment of the error occurring, but I have no choice due to I need to process the packet to get opportunity to interrupt the request on device-dependent error before the new packet arrives.

The protocol assumes the exchanging by several alternating Vendor-IN and Vendor-OUT requests. To simplify the protocol, the Vendor-Write request does not have any useful payload (frankly saying, for the LibUsb 1.2.6.0 Windows Driver I had to pass a dummy payload to proceed the request), but it is not the point.
So, the main point. Time to time the driver performs the next Vendor-IN request without awaiting the previous Vendor-OUT request (Vendor-OUT timeout does not hit), and the device enters the error state because it still processes the previous Vendor-OUT request. Sometimes everything is fine, but sometimes the error occurrs.
What I did: I replaced the Vendor-OUT with the Vendor-IN request (these requests does not any usefull payload so this replacement is possible for me), the the problem vanishes!
Without chaning the main requests sequence, timeouts... I've just replaced the Control-OUT with Control-IN requests, so I have only Vendor-IN requests in the sequence while the all functionality is kept unchanged.

I understand that this issue is to hard to play back, but maybe anybody can help!
Is it a bug?

alexanderplanar, 2018-12-17:
The sequence looks like this:
  [Vendor IN], [Vendor OUT], [Vendor IN] [Vendor IN] [Vendor IN] etc..

[Vendor IN] - in most cases is the "Status request".
[Vendor OUT] - the request to change the device state with some parameters.

So, no I see that I unable to send a data to the device using [Vendor OUT].

By the way, other channel uses Bulk-Request, it works fine.

diabolo38, 2018-12-17:
Sound an issue in your f/w 
If the out complete nicely on host side 
Then sendind a new in or out is legal your device shall handle it or nak the setup if not ready.
Typically you should delay the out status until your not done, in many USB device framework this is not easy or impossible.

To me using control for lengthy slow things is already a bad choice. Use  bulk ep pair instead.

alexanderplanar, 2018-12-17:
@diabolo38, the problem is exactly that the OUT on the host side completes nicely before the device processed the packet. At least it looks like this. I do not understand why the IN-requests works fine, while the IN and OUT requests are processed the same way: first peek the data from endpoint 0, then the preprocessing happens, then clear the buffer (confirm/validating).
The point is that the EP0 in this hardware can receive the data legally even the EP is not empty and contain unconfirmed packet...
By my side, I have no choice, I must process the packet before confirmation to provide legal STATUS stage: I need either confirm the packet, or halt the endpoint by the processing result.
The timeout does not hit, the device processes the request immediatelty, because the request has no real data, the request just changes the internal device status (like device enter the service mode, for example).
As for slow things, I do not use the Control-transfers for that issues anyway, but I use Bulk-transfers instead. The control-transfers I'm talking about are the fast lightweigth service requests

alexanderplanar, 2018-12-17:
It seems I found the root of the evil :-) The problem is in the low-level hardware driver: it able tp catch "empty" interrupts with no data. It clears the IRQ-flag in this situation. Maybe it is wrong behaviour. I made it to not clear the flag and everything is fine! Thanks!

closed 2018-12-17 by alexanderplanar