libusb/libusb #514: Async transfer&&The received data is the last time the device sent it

snikeguo, 2018-12-20:
libusb_fill_bulk_transfer(transfer, handle, 0x1| LIBUSB_ENDPOINT_IN, readbuff, 4096, readcallback, NULL, 0xFFFFFFFF);
	r=libusb_submit_transfer(transfer);	
	int actual_length =0;
	int sendcount = 0;
	while (1)
	{
		
		printf("send Threadid:%d,sendcount %2d\n", GetCurrentThreadId(),++sendcount);
		r = libusb_bulk_transfer(handle, 0x1, writebuff, 1, &actual_length, 1);		
		if (r < 0)
			printf("libusb_bulk_transfer error:%s\n", libusb_error_name(r));
		getchar();
	}

static void readcallback(struct libusb_transfer * transfer)
{
	int i = 0;
	static int cont = 0;
	printf("read ThreadID:%d,actual_length: %2d\n", GetCurrentThreadId(), transfer->actual_length);
	if (transfer->status == LIBUSB_TRANSFER_COMPLETED)
	{
		for (i = 0; i < transfer->actual_length; i++)
		{
			printf("%d,", transfer->buffer[i]);
			if ((i + 1) % 10 == 0)
			{
				printf("\n");
			}
		}
	}
	printf("\n");
	int r = libusb_submit_transfer(transfer);
	if (r < 0)
		printf("libusb_submit_transfer error:%s\n", libusb_error_name(r));
}
The PC sends a  data and the device will reply a data.
but When the PC first sends data, it does not trigger readcallback function. 
When the PC sends data for the second time, it will trigger readcallback, but the received data is sent by the device for the first time.
[image]

snikeguo, 2018-12-20:
The problem has been solved：
async read step : (Pseudo code)
usb_event_poll()
{
	while (1)
	{
		libusb_handle_events_completed();
	}
}
read_callback()
{
	//your code;
	libusb_submit_transfer(read_transfer);
}

init step:
1.libusb_init();
2.create a thread(usb_event_poll);
3.libusb_set_configuration();
4.libusb_claim_interface();
5.read_transfer = libusb_alloc_transfer();
6.libusb_fill_bulk_transfer(read_transfer, read endpoint, read_callback);
7.libusb_submit_transfer(read_transfer);//async read

sync  write step:
libusb_bulk_transfer(handle, 0x1, writebuff, 1, &actual_length, 100);//sync write

mcuee, 2019-01-17:
You may want to ask in libusb-devel mailing list to get better help.

snikeguo, 2019-01-23:
@mcuee The problem has been solved

closed 2019-01-23 by snikeguo

zhangzilong051524, 2019-07-23:
hello, I have a question to ask you when I see your question. You set the cache to 4096 in "libusb_fill_bulk_transfer", but there is not so much data when actually receiving it. When I use it, I have to fill the cache before I enter the callback. Why?