libusb/libusb #518: Opening device resets after a couple of seconds if connected to xhci it works with uhci

czras, 2019-01-08:
I have two laptops.

The first one, 'waggle' is a Lenovo 3000 N200 roughly 12 years old. The other 'bacon' is a Lenovo T460p 2 years old. I have a low speed Polar IR interface which works with N200 but not with the T460p. Libusb version and kernel version does not matter, I tried libusb 1.0.20-22 and a couple of kernels between 4.12-4.18

My test code uses github.com/google/gousb, but also the s710 tool does not work on 'bacon' which is using libusb-compat.

Attaching libusb debug logs with some dmesg and uname output and the lsusb list and tree output.
[bacon.log](https://github.com/libusb/libusb/files/2738372/bacon.log)
[waggle.log](https://github.com/libusb/libusb/files/2738373/waggle.log)
[bacon.lsusb.log](https://github.com/libusb/libusb/files/2738377/bacon.lsusb.log)
[waggle.lsusb.log](https://github.com/libusb/libusb/files/2738378/waggle.lsusb.log)

czras, 2019-01-08:
On bacon as it is seen in the dmesg output, roughly five seconds after trying to open the device, there is a reset which essentially causes the i/o error afterwards (because the device number changes after the reset).

My suspicion is that when opening the device invalid configuration is set. I have found the following on some old mailing list: https://sourceforge.net/p/libusb-win32/mailman/message/3744447/

Alsoattaching verbose lsusb output.
[bacon.lsusb.verbose.log](https://github.com/libusb/libusb/files/2738466/bacon.lsusb.verbose.log)

czras, 2019-01-09:
I had success creating an usbmon log filtered to the relevant devices, see attached.
[bacon.usbmon.filtered.log](https://github.com/libusb/libusb/files/2742448/bacon.usbmon.filtered.log)

czras, 2019-01-09:
taking the same log on the working laptop
[waggle.usbmon.filtered.log](https://github.com/libusb/libusb/files/2742532/waggle.usbmon.filtered.log)

czras, 2019-01-15:
could be some kind of HW issue as with a different laptop (DELL Precision 5530 alias 'stoater') it works
[stoater.lsusb.log](https://github.com/libusb/libusb/files/2761378/stoater.lsusb.log)

mcuee, 2019-01-17:
Not an issue with libusb. You may want to ask in the linux-usb mailing list.

closed 2019-01-17 by mcuee