libusb/libusb #523: libusb master: bluetooth dongles not being recognized on Mac

ricardoquesada, 2019-01-15:
v1.0.21 / v1.0.22 working Ok with bluetooth dongles on Mac.
When using master as of today, it returns `kIOReturnNotReady` ( libusb error is: `-99`) when calling `libusb_set_configuration()`

this can be reproduced by using btstack ( https://github.com/bluekitchen/btstack ) and trying any of the examples from there.

Tested on macOS 10.14.2, using an ASUS BT-400 bluetooth dongle with btstack-master as of today.

hjelmn, 2019-01-15:
Can you do a git bisect and see if you can isolate the commit that broke it for you? I don't have any BT dongles to test with.

ricardoquesada, 2019-01-16:
yep. this one seems to be the culprit:

c14ab5fc4d22749aab9e3534d56012718a0b0f67 is the first bad commit
commit c14ab5fc4d22749aab9e3534d56012718a0b0f67
Author: Nathan Hjelm <hjelmn@lanl.gov>
Date:   Thu Jul 26 12:48:07 2018 -0600

    os/darwin: switch from using ResetDevice to USBDeviceReEnumerate

    In newer versions of macOS the ResetDevice function does nothing. For
    these systems we have to use USBDeviceReEnumerate. This should also
    work for older versions os MacOS X.

hjelmn, 2019-01-16:
Interesting. That commit fixed a very real bug. Not sure if this is a bluestacks bug (but properly handling the renumeration or something). Like the commit says ResetDevice has been a no-op since Apple rewrote the USB stack. Will take some digging to figure out how to fix this.

ricardoquesada, 2019-01-16:
Not familiar with btstack internals. adding @mringwal (btstack author) into the look

ricardoquesada, 2019-01-16:
@hjelmn what you mentioned about being a no-op before, makes sense. if I comment out the call to `libusb_reset_device()`, btstack works ok. but on linux it works ok without comment it out.

mringwal, 2019-01-23:
thanks both for the digging. A bit of context. 

BTstack first scans USB for Bluetooth dongles. If it finds one, it calls libusb_open and then libusb_reset_device. The code in question is here: https://github.com/bluekitchen/btstack/blob/master/platform/libusb/hci_transport_h2_libusb.c#L783

Is this incorrect? I'd like to reset/power-cycle/... the Bluetooth dongle before using it.

Thanks for any hints.

hjelmn, 2019-04-04:
I found the issue. A fix is incoming and will be including in the next release.

PR #551 (2019-04-05): darwin: fix reset device
This commit fixes the backend of reset device to restore the state
of the device if possible. This fixes a bug introduced in
c14ab5fc4d22749aab9e3534d56012718a0b0f67. The previous commit
was necessary due to changes in the system USB stack that
essentially turned the ResetDevice function into a no-op. This
required libusb to move to USBDevuceReEnumerate to effectively
reset the device. The problem is that both the device handle and
libusb devices became invalid. This commit fixes the bug by
waiting for the re-enumeration to complete then 1) checking
whether the descrip [... cut]
commit a0b5d27f (2019-04-05): darwin: fix reset device

This commit fixes the backend of reset device to restore the state
of the device if possible. This fixes a bug introduced in
c14ab5fc4d22749aab9e3534d56012718a0b0f67. The previous commit
was necessary due to changes in the system USB stack that
essentially turned the ResetDevice function into a no-op. This
required libusb to move to USBDevuceReEnumerate to effectively
reset the device. The problem is that both the device handle and
libusb devices became invalid. This commit fixes the bug by
waiting for the re-enumeration to complete then 1) checking
whether the descri [... cut]
closed 2019-04-05 by hjelmn