libusb/libusb #528: segfault on first access after plugging in my device

saveriomiroddi, 2019-01-31:
Hello!

On every first access after plugging in my printer/scanner, I get a segfault, with a corresponding libusb segfault in `dmesg`:

```
[   20.562889] scan-thread[2526]: segfault at 7f2800000054 ip 00007f2836b4c69a sp 00007f28211010b0 error 4 in libusb-1.0.so.0.1.0[7f2836b46000+19000]
```

From the second attempt onwards, the scanner works as expected, without segfault.

If I unplug and plug the scanner back, and try to access it, I get the segfault again (and after, working fine again).

I'm on a Linux Mint 19.1 (which is very close to an Bionic 18.04). I can reproduce this 100% of the times, across many kernel versions (from 4.15 to 4.20), and also on the current `master` version of the library.

dickens, 2020-01-13:
Can you provide a debug log and reproduce this crash?  Configure the library using the `--enable-debug-log` or set `LIBUSB_DEBUG=4` in the environment.

saveriomiroddi, 2020-01-21:
> Can you provide a debug log and reproduce this crash? Configure the library using the `--enable-debug-log` or set `LIBUSB_DEBUG=4` in the environment.

Thanks for answering :smile: 

I'll publish what requested (and/or ask clarification, if required), but it'll take me a while.

saveriomiroddi, 2020-02-24:
Hello! Here's the requested/related information.

Note that I'm on a stock Ubuntu [MATE] 18.04 now, but the behavior is the same.

`dmesg` output:

```
[   39.393882] usblp0: removed
[   39.394701] show_signal_msg: 14 callbacks suppressed
[   39.394703] scan-thread[7484]: segfault at 7f19000000d0 ip 00007f1999f82aca sp 00007f1979c9e0f0 error 4 in libusb-1.0.so.0.1.0[7f1999f7d000+17000]
[   39.394709] Code: 55 53 48 8d 0d 0e a4 00 00 48 8d 15 50 ac 00 00 31 c0 48 89 fb 48 83 ec 08 41 89 f0 31 ff 89 f5 be 04 00 00 00 e8 26 ec ff ff <48> 8b 43 40 8b 40 7c 85 c0 74 23 48 8d 05 c4 11 21 00 48 8b 80 c8
```

Output of `LIBUSB_DEBUG=4 simple-scan`: [debug_simple_scan_output.log](https://github.com/libusb/libusb/files/4243788/debug_simple_scan_output.log)

dickens, 2020-08-18:
Based on the log, this looks like a possible use-after-free caused by the application.  A context was just destroyed and a device was accessed immediately after.  It's not entirely clear if that device belonged to the just-destroyed context.

One other thing is that the log shows is a major imbalance between `libusb_init()` and `libusb_exit()`.  This should be fixed.

saveriomiroddi, 2020-08-18:
> Based on the log, this looks like a possible use-after-free caused by the application. A context was just destroyed and a device was accessed immediately after. It's not entirely clear if that [...]

Thanks for looking into this. The device is unfortunately both with closed source drivers (Samsung C480x) and not supported anymore, but of course, I can provide my hands on :smile:.

Is this error:

> [ 4.955220] [00001d3c] libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/002/001: Permission denied

Something that can be dealt at a higher level (that is, with anything available from a standard distribution), or is it a low level (eg. behavior hardcoded in the binary)?

dickens, 2020-08-18:
> Is this error:
>
>
> Something that can be dealt at a higher level (that is, with anything available from a standard distribution), or is it a low level (eg. behavior hardcoded in the binary)?

This just means that the access permissions on the device node were not set to allow a non-privileged user to access it.  Generally a udev rule or something similar is necessary to adjust permissions.  However I see this message a lot in the log, indicating that the application is trying to open seemingly every device it encounters.  I would therefore expect this message because many devices such as the root hubs and hubs are accessible only to root.

The easy way to address this error is to not open a device which you aren't really interested in.  Both the device descriptor and config descriptor(s) are accessible without needing to open the device, so enough information is usually available to locate the device(s) of interest.

mcuee, 2020-12-07:
Close this for now as it does not seem to be a libusb problem, but rather a Samsung application problem.

closed 2020-12-07 by mcuee