libusb/libusb #534: libusb_open() return -5, LIBUSB_API_VERSION 0x01000107

troies, 2019-02-11:
Recently, I am studying on libusb and want to use libusb to operate the MSC USB device on Win7-x64 platform, I use almost the latest libusb in github, however, I meet an issue that it returns -5 when I invoke the API libusb_open().
The MSC device inserted in my PC could be enumerated, and I could copy file into the MSC device, but I could not open the MSC device with libusb.
If I install the libusb driver by inf-wizard.exe or zadig.exe, I could open the MSC device by libusb API.

Here is my question:
Could I open the MSC device by libusb API with out the libusb driver installed from inf-wizard.exe or zadig.exe?

diabolo38, 2019-02-11:
It may be possible by using usbdk back-end. But wihout spécific driver you can open any device already  use by os

troies, 2019-02-12:
> It may be possible by using usbdk back-end. But without spécific driver you can't open any device already use by os

Thanks for your comments, but I have to use the libusb API, since only in this way can I make minimum change on my code.

troies, 2019-02-12:
I have debugged the libusb code, and I find that when I do not install the driver from zadig.exe, it invokes the composite_open() function, otherwise, it invokes the winusbx_open() function.
Could any one give some explanation for me? Thanks.

patstew, 2019-05-21:
You need to have a libusb compatible driver loaded to use libusb, e.g. winusb, usbdk, libusbK, HID driver. The windows mass storage driver is not compatible with libusb, and libusb cannot change what driver windows uses. You need to either use Zadig, or make your own .inf file that tells windows what driver to load for your device and sign that .inf file yourself. You need to pay for a certificate to sign an .inf file that other people's computers will accept. https://docs.microsoft.com/en-us/windows-hardware/drivers/install/driver-signing

dickens, 2020-01-13:
Going to close this.  As the comments have said, a compatible driver is needed for libusb to use the device on Windows.

closed 2020-01-13 by dickens