libusb/libusb #538: Isochronous Transfers using different backends.

vijaykumarprathipati, 2019-02-19:
Dear Community,

We need some help regarding Isochronous Transfers. We have created an application that sends and receives isochronous packets using libusb api. When we tried with Winusb backend, we are unable to submit transfer(We used Windows 10). When we tried with libusbk backend, isochronous transfers are  working fine if we use fullspeed device, but we are seeing errors(LIBUSB_TRANSFER_ERROR) in the case of highspeed device. This is for both Windows 7 & Windows 10. When we tried with usbdk backend, it works for fullspeed and highspeed devices on Windows 7 and it fails(actual_length in libusb_transfer is showing 0 in the callback - https://github.com/daynix/UsbDk/issues/66) for both on Windows 10.

Can you please let us know which backend needs to be used for Isochronous Transfers for  FullSpeed & HighSpeed devices for platforms Windows 7 & Windows 10 ?

Thanks for all your Help in Advance,

Vijay.

cl-ew, 2019-02-28:
Our team is affected by this issue as well.  Hoping it can get some attention soon :)

mcuee, 2019-04-07:
Try the latest RC to see if that helps.
ISOC transfer is tough to get right.

mcuee, 2021-01-23:
Close this for now. Please reopen with more details.

closed 2021-01-23 by mcuee