libusb/libusb #540: NULL pointer crash during transfer

steven807, 2019-03-14:
libusb version 1.0.1 (commit 2a7372d)
On Mac OS X (Mojave: 10.14.3),
The following crash occurs (in the context of a `libuvc` call to `uvc_get_device_descriptor`):
```
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x58)
  * frame #0: 0x000000010620dc96 libusb-1.0.1.dylib`usbi_handle_transfer_completion [inlined] remove_from_flying_list(transfer=0x0000000107a00a00) at io.c:1444 [opt]
    frame #1: 0x000000010620dc96 libusb-1.0.1.dylib`usbi_handle_transfer_completion(itransfer=0x0000000107a00a00, status=LIBUSB_TRANSFER_CANCELLED) at io.c:1651 [opt]
    frame #2: 0x000000010620eb72 libusb-1.0.1.dylib`handle_events(ctx=0x0000000106407c30, tv=<unavailable>) at io.c:2219 [opt]
    frame #3: 0x000000010620e57c libusb-1.0.1.dylib`libusb_handle_events_timeout_completed(ctx=0x0000000106407c30, tv=<unavailable>, completed=0x00007ffeefbfca94) at io.c:2358 [opt]
    frame #4: 0x000000010620edc6 libusb-1.0.1.dylib`libusb_handle_events_completed(ctx=<unavailable>, completed=<unavailable>) at io.c:2457 [opt]
    frame #5: 0x000000010620f58b libusb-1.0.1.dylib`libusb_control_transfer [inlined] sync_transfer_wait_for_completion(transfer=0x000000010640d068) at sync.c:51 [opt]
    frame #6: 0x000000010620f54c libusb-1.0.1.dylib`libusb_control_transfer(dev_handle=<unavailable>, bmRequestType='\x80', bRequest=<unavailable>, wValue=768, wIndex=<unavailable>, data="p�@\x06\x01", wLength=255, timeout=1000) at sync.c:130 [opt]
    frame #7: 0x000000010620c972 libusb-1.0.1.dylib`libusb_get_string_descriptor_ascii [inlined] libusb_get_string_descriptor(dev_handle=0x0000000106813200, desc_index='\0', langid=0, length=255) at libusb.h:1816 [opt]
    frame #8: 0x000000010620c94e libusb-1.0.1.dylib`libusb_get_string_descriptor_ascii(dev_handle=0x0000000106813200, desc_index='\x02', data="PͿ��", length=64) at descriptor.c:1159 [opt]
```

The libusb_transfer object at that point is:
```
{
  dev_handle = 0x0000000000000000
  flags = '\0'
  endpoint = '\x83'
  type = '\x03'
  timeout = 0
  status = LIBUSB_TRANSFER_COMPLETED
  length = 32
  actual_length = 0
  callback = 0x00000001042be8e0 (libuvc.0.dylib`_uvc_status_callback at device.c:1779)
  user_data = 0x0000000107a00070
  buffer = 0x0000000107a000a0 <no value available>
  num_iso_packets = 0
  iso_packet_desc = {}
}
```

Note that `dev_handle` is NULL.

Log is attached.
[libusb.log](https://github.com/libusb/libusb/files/2968833/libusb.log)

hjelmn, 2019-04-05:
Do you know if this is a regression or is this a long standing issue?

steven807, 2019-04-05:
Sorry, I'm not sure.

hjelmn, 2019-04-05:
Can you test with master? Want to make sure this is still broken after recent fixes.

hjelmn, 2019-04-06:
Is the log from the same run as the backtrace?

hjelmn, 2019-04-06:
Something is really odd because the only way a transfer can have dev_handle == NULL is if the transfer was in flight during close.

steven807, 2019-04-06:
Yup, the log and the backtrace are from the same run.

hjelmn, 2019-04-06:
This is what is odd:

```
[ 0.053575] [00000307] libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
[ 0.053579] [00000307] libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
[ 0.053583] [00000307] libusb: debug [do_close] Removed transfer 0x7fa3b6c04c58 from the in-flight list because device handle 0x7fa3b9800000 closed
```

This transfer (0x7fa3b6c04c58) had its dev_handle set to NULL. The problem is that transfer is the not the same one as above (0x0000000107a00a00). If it is from a different run that would explain it.

steven807, 2019-04-06:
Hmm, not sure what's going on. If I can I'll see if I can remember how I reproduced it, but it's been a few weeks and the details are lost to me right now.

hjelmn, 2019-04-06:
It looks to me like there is a bug that can occur when a device is closed with an outstanding transaction. Will see if I can find it with what you have provided so far.

alxbur, 2019-04-06:
I made some system cleaning, and now got next message while running my routine on terminal. Noted that debugger still shows same spot as a segfault point - it is calloc in core.c
Thanks everybody for the inputs
Alex.
$ ./ve12demo_sw2f 
malloc(): invalid size (unsorted)
Aborted (core dumped)

alxbur, 2019-04-06:
I just tried NULL instead of libusb_context **, as in example - all the same. example works, my routine segfaulted. I'm lost nowhere and have no clue how to proceed further. Any help appreciated.
Alex

hjelmn, 2019-04-07:
@alxbur Think you meant to comment on your bug.

mcuee, 2021-06-05:
https://github.com/libusb/libusb/issues/297
From Chris:
> Closing a device while there are outstanding transfers is 100% an application error. Handling this gracefully has been discussed in #540, #610 and #703.

mcuee, 2021-06-05:
I will close this and please refer to #703 in the future.

closed 2021-06-05 by mcuee

mcuee, 2023-02-06:
1) https://github.com/libusb/libusb/issues/703
2) https://github.com/libusb/libusb/issues/182
3) https://github.com/libusb/libusb/issues/297
4) https://github.com/libusb/libusb/issues/540
5) https://github.com/libusb/libusb/issues/610