libusb/libusb #541: Winusb Isochronous Transfers not Working.

vijaykumarprathipati, 2019-03-19:
Hi,

I tried performing isochronous In and Out Transfers using Winusb and it is throwing error that "Failed to submit urb". The same code is working fine with usbdk backend. The issue is for both FullSpeed and HighSpeed and I am using Windows 10. Can you please help me with this. Below are the samples with which we can recreate the problem.
[isoch_out.txt](https://github.com/libusb/libusb/files/2982519/isoch_out.txt)
[isoch_in.txt](https://github.com/libusb/libusb/files/2982520/isoch_in.txt)

Thanks for all your Help,
Vijay.

mirh, 2019-04-05:
That was only merged some hours ago. 
Not sure what you have been using. 

How's it going now?

mcuee, 2019-04-07:
Please try the latest 1.0.23 rc to see if that helps.

paamand, 2020-01-06:
I have the same problem.
My own code and the isoch_in.txt example both produces the same error, caused by line 2690 of `windows_winusb`:

    WINUSBX_CHECK_API_SUPPORTED(RegisterIsochBuffer);

When debugging the `WinUSBX[sub_api]` seems to refer to a libusbK instance. Could this be a driver conflict?

paamand, 2020-01-06:
There is a conflict in using libusb with a WinUSB device while having libusbK installed.
`winusbx_init(struct libusb_context *ctx)` first defaults to libusbK, and then does not set the `NativeWinUSBOnly_Set(RegisterIsochBuffer)` (among others) for the WinUSB device.

Loading WinUSB solved my problem. Solution could be a flag indicating which to load?

Reason for not using libusbK in my case is because it incorrectly handles latency critical isochronous transfers (handles on a 1ms frame basis?) and reads empty packages for enpoints that uses single package low latency transfers.

mcuee, 2020-01-08:
What do you mean by loading WinUSB? I think you mean using WinUSB driver instead of libusbk,sys and you do not delete libusbk.dll, right?

When you say libusbk incorrectly handles latency critical isochronous transfers, do you mean libusbk.dll or libusbk.sys? I think you mean libusbk.sys, right?

mcuee, 2020-01-08:
BTW, with Windows 7 going to be unsupported, I think libusbk.sys is no longer important. I talked to Travis Robinson (developer of libusbk.sys and libusbk.dll) before and he thinks that the focus should go to WinUSB in the future. 

He and I still think libusbk.dll may have some usages for Windows only projects but more for WinUSB driver. libusbk.dll supports libusbk.sys, libusb0.sys and WinUSB.

mcuee, 2020-01-08:
The whole Windows backend is very complicated, partly due to the support of HID backend, and libusbK.sys and libusb0.sys. Probably it is good to streamline and support WinUSB in the future. But I understand that WinUSB does have some limitations as mentioned in the Wiki.

paamand, 2020-01-08:
@mcuee By 'loading' I mean `h = LoadLibraryA("WinUSB");`. E.g. the DLL.

Actually both libusbK and WinUsb can only handle 1ms frames, since the windows user-space API packages the transfers in the 1ms frames, so my latency critical application (requring <1ms roundtrip, e.g. microframe transfers) seems to not be able to use libusb at all.. :-/
(Can you confirm this?)

I fully agree that focus on WinUSB could be a way forward. However I can too easily get blue-screen crashes with isochronous transfers (I will file another issue on this).
The WinUSB limitations seems to be most severe for older platforms.

I will explore the WinUSB.sys (kernel mode) and usbDk options regarding the microframe latency.

paamand, 2020-01-08:
@mcuee I just tried with WinUSB.sys (e.g. with the device configured and installed as a WinUSB Device). Same results.

Unable to use microframes and transfer frames at 1kHz only. This is very disappointing (but obviously not the fault of libusb) :(

mcuee, 2021-07-05:
@paamand 
If you are using Windows only, especially on Windows 10, you may take a look at new libusbk library which adds the support of isoc transfer with WinUSB (along with better support for libusbk).

It has not been merged but the test results are pretty good.
https://github.com/mcuee/libusbk/tree/WinUSB-ISO
https://github.com/mcuee/libusbk/blob/WinUSB-ISO/BmFW/CYPRESS_FX3/WinUsb_Results.txt

paamand, 2021-07-05:
@mcuee Looks promising, thanks. I cannot see from the test results if you can achieve microframe latency? I was my understanding that the user-space Windows api only allows 1ms frame transfers.

For reference, see my comments to the updated MicrosoftDocs: https://github.com/MicrosoftDocs/sdk-api/pull/34

mcuee, 2021-07-05:
I am not so sure exactly what you mean by micro frame latency. 
You can take a look at Cypress AN86947 (https://www.cypress.com/file/125281/download).

The isoc test results are with the following configuration.
https://github.com/mcuee/libusbk/blob/WinUSB-ISO/BmFW/CYPRESS_FX3/BMFW_IsoSourceSink/cyfxisosrcsink.h
```
/* Maximum packet size for the endpoint. Can take any value from 1 to 1024. */
#define CY_FX_ISO_MAXPKT                (1024)

/* ISO Mult settings for SS and HS operation. */
#define CY_FX_ISO_PKTS                  (3)

/* Burst length in packets supported by the ISO endpoints, when operating in USB 3.0 mode. */
#define CY_FX_ISO_BURST                 (8)

/* Dma buffer size multiplier. As ISO endpoints will only transfer ISO_PKTS * ISO_BURST packets of data in
 * each interval, this can be set to 1 without any performance limitation. */
#define CY_FX_DMA_MULTIPLIER            (1)
```

mcuee, 2021-07-05:
IMHO Windows is not an RTOS so those sub 1ms latency is not really realistic. It is probably the same for Linux and macOS. I understand there are extension to Windows and Linux but I am not so sure whether they really offer sub 1ms latency.

paamand, 2021-07-05:
You are right. Nothing libusb/libusbk can do here.
My case is an edge-case related to low-latency communication. The USB spec allows 1/8 ms "micro frames", and the quote from MicrosoftDocs documentation was added by me after I realized that this part of the USB spec is not supported.

Anyway, great to see support for isoc transfers with libusbk.

mcuee, 2023-01-23:
Closing this old issue as there is no update from the OP.

closed 2023-01-23 by mcuee

rajhlinux, 2025-01-21:
> You are right. Nothing libusb/libusbk can do here. My case is an edge-case related to low-latency communication. The USB spec allows 1/8 ms "micro frames", and the quote from MicrosoftDocs [...]

I read that document, they said WinUSB supports 250 microseconds:

```
_The host controller sends or receives data during reserved periods of time on the bus, are called bus intervals. The unit of bus interval depends on the bus speed. For full speed, it's 1-millisecond frames, for high speed and SuperSpeed, it's 250-microseconds microframes.
```

Reference:
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-set-up-to-use-windows-devices-usb

If it was you because Microsoft had to update their documentation, I have to say I was quite surprised that people in microsoft actually mentions things in microsecond after reading that document.