libusb/libusb #544: USB device discovery not working properly on Dell USB-C docking stations

richardhendricks, 2019-03-26:
When doing some work, our team discovered that USB devices are not properly discovered by libusb on Dell docking stations. 

We have confirmed this issue with multiple device types. We are testing with the listdev.py sample code at https://github.com/vpelletier/python-libusb1/tree/master/examples and libusb-1.0.22 on Windows 10.

USB device not detected by libusb when in the left port (port 3)
[image]

USB device detected by libusb when in the right port (port 4)
[image]

diabolo38, 2019-03-27:
It is likely linked to ASmedia host controler 
The same hapen with TI or NEC (odl driver) XHCI controler
So  far am scared their is  no solution to it :(
Maybe some extra efforts will be put to get this old issue sorted it it affect more commonly used h/w

for further  investigation you will have to provide debug listing 
run again listdev  after setting env var LIBUSB_DEBUG to 4

richardhendricks, 2019-03-27:
Seems to be.

Right slot, working
[right_working.log](https://github.com/libusb/libusb/files/3014388/right_working.log)

Left slot, not working
[left_not_working.log](https://github.com/libusb/libusb/files/3014389/left_not_working.log)

richardhendricks, 2019-03-27:
[image]

diabolo38, 2019-03-28:
If this can help 
in Win7 with Ti XHCi the error is similar 

using A0001 Phone 

On port 3  list fail 
```
[ 0.144000] [00001720] libusb: debug [init_device] (bus: 3, addr: 5, depth: 3, port: 3): 'USB\VID_0553&PID_0608\2.5B0'
[ 0.144000] [00001720] libusb: debug [winusb_get_device_list] found existing device for session [1E40]
[ 0.144000] [00001720] libusb: error [init_device] device 'USB\VID_05C6&PID_6764\B3D44D8A' is no longer connected!
[ 0.145000] [00001720] libusb: debug [winusb_get_device_list] found existing device for session [2F90]
[ 0.145000] [00001720] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.146000] [00001720] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
```

On port 1  it work 
```
[ 0.123000] [00001434] libusb: debug [init_device] (bus: 3, addr: 5, depth: 3, port: 3): 'USB\VID_0553&PID_0608\2.5B0'
[ 0.124000] [00001434] libusb: debug [winusb_get_device_list] found existing device for session [1E40]
[ 0.124000] [00001434] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.126000] [00001434] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.126000] [00001434] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_05C6&PID_6764\B3D44D8A'
[ 0.126000] [00001434] libusb: debug [winusb_get_device_list] found existing device for session [2F90]
```

Mass storage function access  work whatever ports

gnieboer, 2019-07-07:
Can confirm the issue with an Asus X99-A motherboard ASMedia controller.  When plugged into the Intel controller, devices are enumerated by listdevs.  But into the ASMedia connected slots, the devices still appear as expected in Device Manager, but are not included in the listdevs command with libusb 1.0.22.

OS = Windows 10 v1703 and v1903.
Latest ASMedia drivers installed (1.16.35.1)

fridgerator, 2019-10-10:
Any update on this?

dickens, 2020-01-13:
I'm not sure there is much that can be done within libusb to solve this.  The HC driver is either returning an invalid port number when the SPDRP_ADDRESS is being queried or the node connection information is incorrect.  I'm betting it is the former that is the problem.

dickens, 2020-08-18:
Going to close this issue.  Hopefully the HC vendor can provide an updated driver.

closed 2020-08-18 by dickens