libusb/libusb #545: make errors while building libusb

sharma-kullu, 2019-03-28:
$ make check
Making check in libusb
make[1]: Entering directory '/home/alam/libusb/libusb'
  CC       libusb_1_0_la-io.lo
io.c: In function ‘usbi_io_init’:
io.c:1164:2: error: implicit declaration of function ‘close’; did you mean ‘_lclose’? [-Werror=implicit-function-declaration]
  close(ctx->timerfd);
  ^~~~~
  _lclose
In file included from io.c:37:0:
io.c: In function ‘disarm_timerfd’:
io.c:1319:11: warning: zero-length gnu_printf format string [-Wformat-zero-length]
  usbi_dbg("");
           ^
libusbi.h:236:71: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __FUNCTION__, __VA_ARGS__)
                                                                       ^~~~~~~~~~~
io.c:1319:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg("");
  ^
io.c: In function ‘libusb_interrupt_event_handler’:
io.c:1897:11: warning: zero-length gnu_printf format string [-Wformat-zero-length]
  usbi_dbg("");
           ^
libusbi.h:236:71: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __FUNCTION__, __VA_ARGS__)
                                                                       ^~~~~~~~~~~
io.c:1897:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg("");
  ^
cc1: some warnings being treated as errors
make[1]: *** [Makefile:664: libusb_1_0_la-io.lo] Error 1
make[1]: Leaving directory '/home/alam/libusb/libusb'
make: *** [Makefile:476: check-recursive] Error 1

LudovicRousseau, 2019-03-28:
What operating system?
What compiler?

sharma-kullu, 2019-03-29:
Operating system : cygwin64 on windows 10 
Compiler : gcc for cygwin ( standard package)

sharma-kullu, 2019-03-29:
Cygwin also has package for libusb1.0 but it's too is not working when I am using it with find_library(LIBRARY ) in cmake. I guess somehow it's not getting installed when done using cygwin packages.. Therefore tried installing using git, but found above errors 
Steps taken :
Cd libusb 
./bootstrap 
make && make install

sharma-kullu, 2019-03-31:
`/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0xfa): undefined reference to `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0xfa): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x70b): undefined reference to `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x70b): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x752): undefined reference to `__imp___iob_func'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x752): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___iob_func'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x7b5): undefined reference to `__imp___iob_func'
/bin/libusb-1.0.a(libusb_1_0_la-core.o):core.c:(.text+0x7b5): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp___iob_func'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x3b9): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x3b9): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x5f4): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x5f4): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xb88): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xb88): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xbc7): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xbc7): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xcd6): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0xcd6): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x1565): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-io.o):io.c:(.text+0x1565): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0xf8): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0xf8): additional relocation overflows omitted from the output
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x126): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x432): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x4ee): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x52a): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x594): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x5cf): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x5f3): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x6c8): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x70f): undefined reference to `__imp__errno'
/bin/libusb-1.0.a(libusb_1_0_la-poll_windows.o):poll_windows.c:(.text+0x739): undefined reference to `_assert'
/bin/libusb-1.0.a(libusb_1_0_la-windows_nt_common.o):windows_nt_common.c:(.text+0x921): undefined reference to `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-windows_nt_common.o):windows_nt_common.c:(.text+0x12d3): undefined reference to `__imp__beginthreadex'
/bin/libusb-1.0.a(libusb_1_0_la-windows_nt_common.o):windows_nt_common.c:(.text+0x1a0e): undefined reference to `__imp__strdup'
/bin/libusb-1.0.a(libusb_1_0_la-windows_usbdk.o):windows_usbdk.c:(.text+0x10d1): undefined reference to `__ms_vsnprintf'
/bin/libusb-1.0.a(libusb_1_0_la-windows_winusb.o):windows_winusb.c:(.text+0x5e22): undefined reference to `__imp__stricmp'
/bin/libusb-1.0.a(libusb_1_0_la-windows_winusb.o):windows_winusb.c:(.text+0x5ea0): undefined reference to `__imp__strdup'
/bin/libusb-1.0.a(libusb_1_0_la-windows_winusb.o):windows_winusb.c:(.text+0x6235): undefined reference to `__imp__strdup'
/bin/libusb-1.0.a(libusb_1_0_la-windows_winusb.o):windows_winusb.c:(.text+0x6c0f): undefined reference to `__imp__strdup'
/bin/libusb-1.0.a(libusb_1_0_la-windows_winusb.o):windows_winusb.c:(.text+0x6ed1): undefined reference to `__imp__strdup'
collect2: error: ld returned 1 exit status
`
Error list I get when I install libusb using CYGWIN standard packages.
Any point I am missing here?

LudovicRousseau, 2019-03-31:
It is strange some functions are not declared. Like `close()`:
```
io.c:1164:2: error: implicit declaration of function ‘close’; did you mean ‘_lclose’? [-Werror=implicit-function-declaration]
```
I guess your compiler installation is missing some header files. `close()` is a very classic function for POSIX systems.

I do not use Windows or Cygwin so I can't help much.

mcuee, 2019-04-07:
I think for Cygwin you have to use their own build. If Cygwin package does not work you should get support from them.

dickens, 2020-01-13:
This is a toolchain problem.  Windows doesn't support timerfd so the fact that autoconf detected support for this means it is broken.

closed 2020-01-13 by dickens