libusb/libusb #547: Feature Request: Add libusb_open_device_with_vid_pid(libusb_context *ctx, uint16_t vendor_id, uint16_t product_id,uint16_t deviceIndex) function

snikeguo, 2019-04-02:
libusb_open_device_with_vid_pid(ctx, usb_vid, usb_pid,0);
libusb_open_device_with_vid_pid(ctx, usb_vid, usb_pid,1);
This is more convenient.what do you think？

mcuee, 2019-04-07:
The problem is what is deviceIndex defined.

hjelmn, 2019-04-10:
I agree. There will be some degree of nondeterminism with the ordering of devices in the device list. You can always get the device list and run through it yourself.

dickens, 2020-01-13:
Going to close this for now.  The `libusb_open_device_with_vid_pid()` function is really meant for applications that want a simplified way of opening a specific device when there will only be one such device connected.  As mentioned, there is too much non-determinism in how the devices are enumerated that the value of the proposed function is questionable.

closed 2020-01-13 by dickens