libusb/libusb #549: macOS aborts all transactions on libusb_cancel_transfer

najiji, 2019-04-04:
We have libusb opening 4 transfers on the debug channel, that we always immediately resubmit on the callback. 

If another thread calls `libusb_cancel_transfer` on one of the transfers, macOS seems to abort all other running transfers as well, and then hang for about 5s. My application (still assuming that the other transfers haven't been closed) tries to close the other transfers. Invoking `libusb_cancel_transfer` then always induces a lag of about 5s.

Why is macOS aborting all transfers on aborting one? Is there a bug in my code or might that be a problem in libusb?

I am running macOS 10.14.4

```
[ 0.016103] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7f8c7be3fc78
[ 0.016111] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7f8c7be3fc78
[ 0.016114] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.016116] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.016327] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7f8c7be3e1e8
[ 0.016348] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7f8c7be3e1e8
[ 0.016352] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.016356] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.016405] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7f8c7be3e2e8
[ 0.016412] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7f8c7be3e2e8
[ 0.016415] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.016418] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.016458] [00000307] libusb: debug [libusb_alloc_transfer] transfer 0x7f8c7be3e3e8
[ 0.016465] [00000307] libusb: debug [libusb_submit_transfer] transfer 0x7f8c7be3e3e8
[ 0.016468] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.016472] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.016565] [00000307] libusb: debug [libusb_cancel_transfer] transfer 0x7f8c7be3fc78
[ 0.016577] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.016581] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.016583] [00000307] libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 1
[ 0.016702] [00000307] libusb: debug [darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
[ 0.016780] [00002147] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.016830] [00002147] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.016842] [00002147] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.016854] [00002147] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.016845] [00005203] libusb: debug [handle_events] poll() returned 1
[ 0.016883] [00005203] libusb: debug [handle_events] caught a fish on the event pipe
[ 0.016888] [00005203] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[ 5.043228] [00000307] libusb: error [libusb_cancel_transfer] cancel transfer failed error -7
[ 5.043267] [00000307] libusb: debug [libusb_cancel_transfer] transfer 0x7f8c7be3e1e8
[ 5.043274] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 5.043278] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 5.043282] [00000307] libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 1
[ 5.043281] [00005203] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f8c7be3fc78 has callback 0x10725a410
[ 5.043306] [00000307] libusb: debug [darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
[ 5.043308] [00005203] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[10.092191] [00000307] libusb: error [libusb_cancel_transfer] cancel transfer failed error -7
[10.092227] [00000307] libusb: debug [libusb_cancel_transfer] transfer 0x7f8c7be3e2e8
[10.092233] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[10.092237] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[10.092241] [00000307] libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 1
[10.092264] [00000307] libusb: debug [darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
[10.092279] [00005203] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f8c7be3e1e8 has callback 0x10725a410
[10.092296] [00005203] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[12.544380] [00000307] libusb: error [libusb_cancel_transfer] cancel transfer failed error -99
[12.544453] [00000307] libusb: debug [libusb_cancel_transfer] transfer 0x7f8c7be3e3e8
[12.544460] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[12.544464] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[12.544468] [00000307] libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 1
[12.544468] [00005203] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f8c7be3e2e8 has callback 0x10725a410
[12.544483] [00005203] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[12.544526] [00000307] libusb: debug [darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
```

hjelmn, 2019-04-04:
I believe this is documented somewhere (I know it has come up before). It is a limitation of the OS interface. We can only cancel all transfers.

llongi, 2019-04-04:
Hi. I'm the original author of the application above. The same code calling cancel_transfer() multiple times on OS X has been working fine for years. Just recently we've hit these 5 second timeouts/delays on the cancel calls after the first one. We tried libusb 1.0.21, 22 and master. Did something change in OS X? Or was this known and we just never hit it?
So the fix is to just issue 1 cancel and wait for all the transfer callbacks to be processed on OS X?
Thanks for your help. I couldn't find anything in the docs at a quick glance.

hjelmn, 2019-04-04:
Not sure of the delay is new. Can take a look and see what I find.

hjelmn, 2019-04-05:
@llongi Is this now fixed on your end? I will keep this bug open while I try and determine if there is something wrong in the darwin backend.

najiji, 2019-04-05:
Unfortunately, this does not solve the issue. Calling `libusb_cancel_transfer` does indeed cancel all transfers, but always times out after 5 seconds, even on the first call

```
[ 0.032636] [00000307] libusb: debug [libusb_cancel_transfer] transfer 0x7fbf41d3e6e8
[ 0.032643] [00000307] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.032646] [00000307] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
[ 0.032648] [00000307] libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 1
[ 0.032691] [00000307] libusb: debug [darwin_abort_transfers] calling clear pipe stall to clear the data toggle bit
[ 0.032702] [00000f1f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.032711] [00000f1f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.032715] [00000f1f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.032716] [00004d03] libusb: debug [handle_events] poll() returned 1
[ 0.032719] [00000f1f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.032720] [00004d03] libusb: debug [handle_events] caught a fish on the event pipe
[ 0.032725] [00004d03] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[ 5.096831] [00000307] libusb: error [libusb_cancel_transfer] cancel transfer failed error -7
[ 5.096908] [00004d03] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fbf41d3e6e8 has callback 0x10a255410
[ 5.096922] [00004d03] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[ 5.096928] [00004d03] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fbf41d3e7e8 has callback 0x10a255410
[ 5.096932] [00004d03] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[ 5.096936] [00004d03] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fbf41d3e8e8 has callback 0x10a255410
[ 5.096939] [00004d03] libusb: debug [darwin_handle_transfer_completion] handling interrupt completion with kernel status -536870165
[ 5.096943] [00004d03] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fbf41d3ea28 has callback 0x10a255410
[ 5.096953] [00004d03] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 5.096957] [00004d03] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 5.096961] [00004d03] libusb: debug [handle_events] poll() 1 fds with timeout in 10ms
[ 5.098182] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7fbf41d3e6e8
[ 5.098240] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7fbf41d3e7e8
[ 5.098249] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7fbf41d3e8e8
[ 5.098255] [00000307] libusb: debug [libusb_free_transfer] transfer 0x7fbf41d3ea28
[ 5.109575] [00004d03] libusb: debug [handle_events] poll() returned 0
```

hjelmn, 2019-04-05:
Can you try the following patch:

```patch
diff --git a/libusb/os/darwin_usb.c b/libusb/os/darwin_usb.c
index dd391db..ef79739 100644
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -1974,13 +1974,12 @@ static int darwin_abort_transfers (struct usbi_transfer *itransfer) {
 #if InterfaceVersion >= 550
   if (LIBUSB_TRANSFER_TYPE_BULK_STREAM == transfer->type)
     (*(cInterface->interface))->AbortStreamsPipe (cInterface->interface, pipeRef, itransfer->stream_id);
-  else
 #endif
-    (*(cInterface->interface))->AbortPipe (cInterface->interface, pipeRef);
 
   usbi_dbg ("calling clear pipe stall to clear the data toggle bit");
 
-  /* newer versions of darwin support clearing additional bits on the device's endpoint */
+  /* newer versions of darwin support clearing additional bits on the device's endpoint. this call will
+     also cancel all outstanding transactions with status kIOUSBTransactionReturned */
   kresult = (*(cInterface->interface))->ClearPipeStallBothEnds(cInterface->interface, pipeRef);
 
   return darwin_to_libusb (kresult);
@@ -2045,6 +2044,7 @@ static enum libusb_transfer_status darwin_transfer_status (struct usbi_transfer
   case kIOReturnSuccess:
     return LIBUSB_TRANSFER_COMPLETED;
   case kIOReturnAborted:
+  case kIOUSBTransactionReturned:
     return LIBUSB_TRANSFER_CANCELLED;
   case kIOUSBPipeStalled:
     usbi_dbg ("transfer error: pipe is stalled");
```

The ClearPipeStallBothEnds call also acts as an abort so calling AbortPipe may have been a little heavy handed.

llongi, 2019-04-05:
Thank you for the patch. It does not fix the 5s timeout, but we've tracked down the issue to a problem in our firmware for that particular device. By looking at the USB traffic with Wireshark we noticed that a control transfer CLEAR FEATURE request was being sent with ENDPOINT HALT, and the response timed out (5s is the USB spec timeout iirc). Turns out on that particular one code path, we weren't Ack'ing the control transfer itself, just clearing the stall on the endpoint to be halted.
Somehow this works fine on Linux and Windows, but OS X seems to be strict and wait for a response.
Thank you for your time and the great support. I think this can be closed now.

hjelmn, 2019-04-05:
Alright. Thanks for the update.

closed 2019-04-05 by hjelmn